open-mmlab/OpenPCDet #857: How to get the model graph

makeradventures, 2022-03-10:
Hello, 
I have a very specific problem I have been struggling with for some time now: Given the architecture of OpenPCDet I am confused how to get the graph of the model! 
I tried following those instructions: [stackoverflow](https://stackoverflow.com/questions/52468956/how-do-i-visualize-a-net-in-pytorch)
but all of them fail, because they expect as an input for the forward method something like a torch tensor, and can not take the given batch as input! For example pasting this into the demo.py file : `make_dot(pred_dicts, params=dict(model.named_parameters()))` gives the error:
[image]
In all of the described methods in the post similar problems appear!
Anyone having experience how to get the graph of the model, without having to change to much code?

Thanks in advance!

makeradventures, 2022-03-10:
I solved this problem by converting frame_id and batch_size to a torch.tensor:
```
    with torch.no_grad():
        for idx, data_dict in enumerate(demo_dataset):
            logger.info(f'Visualized sample index: \t{idx + 1}')
            data_dict = demo_dataset.collate_batch([data_dict]) 
            print("converting frame id, batch size ...")
            data_dict['frame_id']=torch.tensor(data_dict['frame_id'])
            data_dict['batch_size']=torch.tensor(data_dict['batch_size'])   
            load_data_to_gpu(data_dict)
            #trying to get the model trace:
            model_trace = torch.jit.trace(model.forward, data_dict)   
            pred_dicts, _ = model.forward(data_dict)
            V.draw_scenes(
                points=data_dict['points'][:, 1:], ref_boxes=pred_dicts[0]['pred_boxes'],
                ref_scores=pred_dicts[0]['pred_scores'], ref_labels=pred_dicts[0]['pred_labels']
            )
            if not OPEN3D_FLAG:
                mlab.show(stop=True)
```

but now I get the untraceable error Segmentation fault (core dumped):
```
/usr/local/lib/python3.7/dist-packages/spconv/pytorch/core.py:149: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  assert len(spatial_shape) == ndim, "spatial shape must equal to ndim"
/usr/local/lib/python3.7/dist-packages/spconv/pytorch/core.py:151: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  assert batch_size > 0
/usr/local/lib/python3.7/dist-packages/spconv/pytorch/core.py:160: TracerWarning: torch.Tensor results are registered as constants in the trace. You can safely ignore this warning if you use this function to create tensors out of constant variables that would be the same every time you call this function. In any other case, this might cause the trace to be incorrect.
  grid = torch.Tensor()  # empty tensor
/usr/local/lib/python3.7/dist-packages/spconv/pytorch/conv.py:187: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  1] == self.in_channels, "channel size mismatch"
/usr/local/lib/python3.7/dist-packages/spconv/pytorch/modules.py:140: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  if input.indices.shape[0] != 0:
/usr/local/lib/python3.7/dist-packages/spconv/pytorch/conv.py:480: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  if inp.indices.shape[0] != datas.indices.shape[0]:
/usr/local/lib/python3.7/dist-packages/spconv/pytorch/ops.py:428: TracerWarning: torch.Tensor results are registered as constants in the trace. You can safely ignore this warning if you use this function to create tensors out of constant variables that would be the same every time you call this function. In any other case, this might cause the trace to be incorrect.
  pair_mask_bwd = torch.Tensor()
/usr/local/lib/python3.7/dist-packages/spconv/pytorch/ops.py:477: TracerWarning: torch.Tensor results are registered as constants in the trace. You can safely ignore this warning if you use this function to create tensors out of constant variables that would be the same every time you call this function. In any other case, this might cause the trace to be incorrect.
  mask_argsort_bwd = torch.Tensor()
/home/docker/data/OpenPCDet/pcdet/models/backbones_2d/base_bev_backbone.py:95: TracerWarning: Converting a tensor to a Python integer might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  stride = int(spatial_features.shape[2] / x.shape[2])
/home/docker/data/OpenPCDet/pcdet/models/detectors/detector3d_template.py:214: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  assert cls_preds.shape[1] in [1, self.num_class]
/home/docker/data/OpenPCDet/pcdet/models/model_utils/model_nms_utils.py:14: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  if box_scores.shape[0] > 0:
[... cut ...]
  box_scores_nms, indices = torch.topk(box_scores, k=min(nms_config.NMS_PRE_MAXSIZE, box_scores.shape[0]))
/home/docker/data/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_utils.py:91: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  assert boxes.shape[1] == 7
Segmentation fault (core dumped)

```

makeradventures, 2022-03-10:
There may be some connection to #828!

makeradventures, 2022-03-10:
Maybe it is just a bug and one can do nothing about it:
https://github.com/pytorch/pytorch/issues/19349

I think I give up, but I would really appreciate it, if someone could tell me how to visualize the graph of an OpenPCDet model!

closed 2022-04-25 by github-actions[bot]

gzchenjiajun, 2022-05-31:
same problem ..
IndexError: slice() cannot be applied to a 0-dim tensor.
how to solve