open-mmlab/OpenPCDet #858: Lyft Files does not match documentation

OrcunCanDeniz, 2022-03-10:
I downloaded and extracted the Lyft Dataset but instead of `data & maps & images & lidar & train_lidar` folders there were `LICENSE.txt   test_data    test_lidar  train_data    train_lidar
feedback.txt  test_images  test_maps   train_images  train_maps`. How should i proceed from now.

jihanyang, 2022-03-10:
```
OpenPCDet
├── data
│   ├── lyft
│   │   │── ImageSets
│   │   │── trainval
│   │   │   │── data & maps & images & lidar & train_lidar
├── pcdet
├── tools
```
Here, you can first build folder `trainval`, and move `train_data, train_lidar, train_images, train_maps` to `trainval/` and rename them to 'data, lidar, images, maps'. Notice that train_lidar is a softlink for lidar, and just to suit lyft api.

OrcunCanDeniz, 2022-03-10:
Wont this discard all test part of the dataset?

jihanyang, 2022-03-11:
Move test data to a new folder named `test/` is enough. I only mention training and validation here. Notice that our code also generate test info.

OrcunCanDeniz, 2022-03-11:
I did;

    OpenPCDet
    ├── data
    │   ├── lyft
    │   │   │── ImageSets
    │   │   │── trainval
    │   │   │   │── data & maps & images & lidar & train_lidar
    │   │   │── test
    │   │   │   │── data & maps & images & lidar & test_lidar

But info generation did not find `/test` dir. Just processed `/trainval`. It looked liked it was just expecting trainval too. There was a prompt like found 180 samples of 180 samples. Script worked as it was designed but is this the best way. It just skipped 180 samples from test set.

jihanyang, 2022-03-12:
I PR some modification to make this part more understandable. You can refer to the latest GETTING_SARTED.md. I want to highlight that since some path have been fixed in lyft_dataset_sdk, I design such complex directory structure to avoid the modification on lyft_dataset_sdk. If you find simpler way to do so, feel free to let us know.

daeunni, 2022-06-13:
Hi, where is **ImageSets** forder ?

I downloaded dataset here.(https://level-5.global/download/ )