open-mmlab/OpenPCDet #860: Custom dataset training error.

yun9993, 2022-03-11:
keep calling __getitem__.
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data
    return self.__getitem__(new_index)
  File "/nas/lyp/code/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 441, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)   #################################
  File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 127, in prepare_data
    data_dict = self.data_augmentor.forward(
  File "/nas/lyp/code/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 240, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "/nas/lyp/code/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 49, in random_world_flip
    gt_boxes, points = getattr(augmentor_utils, 'random_flip_along_%s' % cur_axis)(
  File "/nas/lyp/code/tools/../pcdet/datasets/augmentor/augmentor_utils.py", line 15, in random_flip_along_x
    enable = np.random.choice([False, True], replace=False, p=[0.5, 0.5])
  File "mtrand.pyx", line 978, in numpy.random.mtrand.RandomState.choice
  File "<__array_function__ internals>", line 5, in unique
  File "/root/anaconda3/envs/openpcdet/lib/python3.8/site-packages/numpy/lib/arraysetops.py", line 262, in unique
    ret = _unique1d(ar, return_index, return_inverse, return_counts)
  File "/root/anaconda3/envs/openpcdet/lib/python3.8/site-packages/numpy/lib/arraysetops.py", line 315, in _unique1d
    ar = np.asanyarray(ar).flatten()
  File "/root/anaconda3/envs/openpcdet/lib/python3.8/site-packages/numpy/core/_asarray.py", line 171, in asanyarray
    return array(a, dtype, copy=False, order=order, subok=True)
RecursionError: maximum recursion depth exceeded while calling a Python object

SaMeEr9597, 2022-04-22:
> keep calling **getitem**. File "/nas/lyp/code/tools/../pcdet/datasets/dataset.py", line 155, in prepare_data return self.**getitem**(new_index) File [...]

Hi, were you able to solve this, I am having the same issue when training custom dataset

ccsself, 2022-05-01:
i met  the same error as yours   have u solved it ?

SaMeEr9597, 2022-05-01:
Nope

closed 2022-06-15 by github-actions[bot]