open-mmlab/OpenPCDet #861: Not very good at detecting nearby vehicles.

Leozyc-waseda, 2022-03-11:
I am trying to use the PV-RCNN model trained on pandaset to test it on my private dataset,

1.I converted my dataset's coordinate system to Normative coordinates are:
            # - x pointing forward
            # - y pointings to the left
            # - z pointing to the top
2. pv_rcnn.yaml from kitti pv_rcnn.yaml，modified
`_BASE_CONFIG_: cfgs/dataset_configs/pandaset_dataset.yaml`
3. In **pandaset_dataset.yaml** I modified
```
POINT_CLOUD_RANGE: [-70, -40, -2, 70, 40, 0] # xmin, ymin, zmin, xmax, ymax, zmax
 VOXEL_SIZE: [0.05, 0.05, 0.05]
```
followed
https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-679190936

You can see that at close range, there are two cars that are not detected, and some strange results are output (yellow and cyan)
[image]

As MartinHahner mentioned in this issue, LiDAR models do not transfer well from one dataset to another one.
https://github.com/open-mmlab/OpenPCDet/issues/599#issuecomment-886656943

But here also mentioned the paper [Train in Germany, Test in The USA: Making 3D Object Detectors Generalize](https://arxiv.org/abs/2005.08139), I found that the accuracy should be quite good in the case of close range.
https://github.com/open-mmlab/OpenPCDet/issues/599#issuecomment-888231775

Do you have any suggestions to improve the problem of not being able to detect cars at close range?

 1.Do I need to retrain the model with my dataset height?
The lidar of pandaset is close to 2m off the ground, and my data is almost around 2.3m
 2.Or are there any parameters I need to adjust carefully?
**Such as POINT_CLOUD_RANGE, VOXEL_SIZE**

jihanyang, 2022-03-11:
I think our work [ST3D](https://github.com/CVMI-Lab/ST3D) can partly help this issue, especially for the 1st problem, which can be solved by just shifting the point cloud on z-axis.

Leozyc-waseda, 2022-03-11:
@jihanyang 
Thank you for your reply.
I will test my data with ST3D

BTW, 
> which can be solved by just shifting the point cloud on z-axis.
Is there anything I need to pay attention to about adjusting the point cloud z-axis?
If the lidar point is based on the ground 0m, or based on the actual height of 2.052m, how to adjust it when adjusting?

jihanyang, 2022-03-12:
You can do similar shifting as 
https://github.com/CVMI-Lab/ST3D/blob/812b64491c3a67a1b578fe46c375fcd0ed2c4305/tools/cfgs/da-waymo-kitti_models/secondiou/secondiou_old_anchor.yaml#L13

Leozyc-waseda, 2022-03-14:
Thank you jihanyang!

closed 2022-03-16 by Leozyc-waseda

jihanyang, 2022-05-31:
Can you highlight the car that you cannot detect?

Leozyc-waseda, 2022-05-31:
Image
[image]

If the score value is 0.1, there will be a lot of FP.
[image]

If the score value is 0.7, there will be a lot of car cannot detect.
[image]

jihanyang, 2022-05-31:
Could you provide how does adaptition model perform on this frame, such as pre-trained waymo?