open-mmlab/OpenPCDet #865: Question about the NMS for waymo dataset

LZDSJTU, 2022-03-13:
I find that the NMS threshold for the waymo dataset is 0.7. I run demo.py and found a lot of boxes overlapping. I don't think it's normal. Also, the final result is much lower than the reported. I do not know if I do sth wrongly.

LZDSJTU, 2022-03-13:
[image]

LZDSJTU, 2022-03-13:
[image]

LZDSJTU, 2022-03-13:
The model is pointpillar_1x, and I do not modify the official code.

MartinHahner, 2022-03-22:
> I find that the NMS threshold for the waymo dataset is 0.7. I run demo.py and found a lot of boxes overlapping. I don't think it's normal. Also, the final result is much lower than the reported. I do not know if I do sth wrongly.

Probably the combination of `pointpillar_1x` and `demo.py` does not execute any NMS, so you could call NMS yourself in [demo.py](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/demo.py) before these lines are executed:

https://github.com/open-mmlab/OpenPCDet/blob/a35f429768b3aa6ea2e7cd0a6452e1fcb7671c8d/tools/demo.py#L100-L103

NMS is provided here:

https://github.com/open-mmlab/OpenPCDet/blob/a35f429768b3aa6ea2e7cd0a6452e1fcb7671c8d/pcdet/ops/iou3d_nms/iou3d_nms_utils.py#L102-L116

KickCellarDoor, 2022-04-10:
The default config uses a very high NMS threshold (0.7) for Waymo, which means only pred boxes with overlap iou larger than 0.7 will be filtered. You may change the NMS threshold to 0.01 and try again.

closed 2022-05-25 by github-actions[bot]

Tartisan, 2022-07-14:
Is it solved? I met the same problem