open-mmlab/OpenPCDet #866: Can we use spconv 2.x with pytorch 1.1 ?

SaMeEr9597, 2022-03-13:
Can we use spconv 2.x with pytorch 1.1 ? or should we should we shift to another pytroch version

jihanyang, 2022-03-14:
The main issue is that spconv2.x can only work on CUDA >= 10.2. You can refer spconv for more details

closed 2022-03-14 by jihanyang

SaMeEr9597, 2022-03-14:
> The main issue is that spconv2.x can only work on CUDA >= 10.2. You can refer spconv for more details

I'm currently using this version, so it should be fine right ? 

[image]

pip3 install torch==1.10.0+cu113 torchvision==0.11.1+cu113 torchaudio==0.10.0+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html