open-mmlab/OpenPCDet #867: python3.8 numpy==1.21.5 numba==0.55.1: ImportError: numpy.core.multiarray failed to import

OrangeSodahub, 2022-03-14:
```
$ python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml     --ckpt pv_rcnn_8369.pth     --data_path data/kitti/training/velodyne/000000.bin
RuntimeError: module compiled against API version 0xf but this version of numpy is 0xe
Traceback (most recent call last):
  File "demo.py", line 18, in <module>
    from pcdet.datasets import DatasetTemplate
  File "/home/zonlin/IPP/OpenPCDet/pcdet/datasets/__init__.py", line 7, in <module>
    from .dataset import DatasetTemplate
  File "/home/zonlin/IPP/OpenPCDet/pcdet/datasets/dataset.py", line 14, in <module>
    from augmentor.data_augmentor import DataAugmentor
  File "/home/zonlin/IPP/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 11, in <module>
    import augmentor_utils, database_sampler
  File "/home/zonlin/IPP/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 6, in <module>
    import SharedArray
ImportError: numpy.core.multiarray failed to import
```
When I installed numpy==1.21.5 then it tells me `numpy.core.multiarray failed to import`
When I upgraded numpy to 1.22 then it tells me `numba requires numpy<1.22`
How can I solve this? Please.......

daeunni, 2022-03-22:
> > Hey, I'm not sure this will solve the issue but I am using `numba 0.54.1` and `numpy 1.19.2` and I have no such problem on my end, so maybe try these versions. Hope it will work !
> 
> I have solved this problem. That's degrade the 'SharedArray' package

hey, what's your 'SharedArray' package version? i have same preoblem with you

OrangeSodahub, 2022-03-22:
> 

Hello, You should degrade the 'SharedArray' package one version by one version, try all the versions. There aren't too many choices. Soon it will works. Like from the latest version, to older one, to more older one...

OrangeSodahub, 2022-03-22:
> hey, what's your 'SharedArray' package version? i have same preoblem with you

Hello, You should degrade the 'SharedArray' package one version by one version, try all the versions. There aren't too many choices. Soon it will works. Like from the latest version, to older one, to more older one..

daeunni, 2022-03-22:
oh, i solved it. thanks XD
my downgraded version is **sharedarray==3.1.0** (i want someone to see this and get help)

saurtone, 2022-05-02:
> Hey, I'm not sure this will solve the issue but I am using `numba 0.54.1` and `numpy 1.19.2` and I have no such problem on my end, so maybe try these versions. Hope it will work !

how about your CUDA version？

OrangeSodahub, 2022-05-02:
> > Hey, I'm not sure this will solve the issue but I am using `numba 0.54.1` and `numpy 1.19.2` and I have no such problem on my end, so maybe try these versions. Hope it will work !
> 
> how about your CUDA version？

This problem has been solved.

saurtone, 2022-05-02:
邮件我已收到。辛苦了。

closed 2022-06-17 by github-actions[bot]