open-mmlab/OpenPCDet #868: explanation for some parameters in dataset and model's yaml file

YCAyca, 2022-03-14:
Hello, there are some parameters that I dont understand and maybe a simple documentation would be useful for everyone. Spesifically I would like to ask what is exactly 

PREPARE: {
             filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5'],
             filter_by_difficulty: [-1],
          }

SAMPLE_GROUPS: ['Car:20','Pedestrian:15', 'Cyclist:15']

and USE_ROAD_PLANE: True

parts in a dataset's yaml (eg kitti_dataset.yaml) file??

or 

MULTI_CLASSES_NMS = False
NMS_PRE_MAXSIZE: 4096
NMS_POST_MAXSIZE: 500

in a models yaml file like second.yaml
I know the logic of NMS but these 3 parameters I couldnt understand their functions

YCAyca, 2022-03-15:
Thank you for your answer!

closed 2022-03-15 by YCAyca

MartinHahner, 2022-03-22:
> filter_by_difficulty: [-1]
this line means: filter annotations with "unknown" difficulty