open-mmlab/OpenPCDet #869: how to show the result of  demo.py in docker ? Thanks

MiaoRain, 2022-03-15:
how to show the result of  demo.py in docker? Thanks

LouisSF, 2022-03-15:
Hey, I don't think this is recommended, you should probably use Open3D to save results as images to a shared folder with the main server (not Docker). You can then visualize them out of the container. 
If you really must visualize them within Docker, maybe look at references such as this [StackOverflow issue](https://stackoverflow.com/questions/16296753/can-you-run-gui-applications-in-a-linux-docker-container) or other [tutorials](https://www.cloudsavvyit.com/10520/how-to-run-gui-applications-in-a-docker-container/)

MiaoRain, 2022-03-15:
> Hey, I don't think this is recommended, you should probably use Open3D to save results as images to a shared folder with the main server (not Docker). You can then visualize them out of the [...]

Thanks a lot!

closed 2022-04-29 by github-actions[bot]

Zishuo-9, 2022-09-27:
How to save the results?