open-mmlab/OpenPCDet #871: __getattr__    raise AttributeError：I ran into this problem when I wanted to train custom data set

969682997, 2022-03-15:
My task was to train the custom dataset, which only contained point cloud data, so I adapted my dataset by imitating KITTI's relevant files. I have tower_infos_*.pkl etc file. __getattr__ occurs when I try to run the 'python my_train.py --cfg_file CFGS /tower_models/my_pointpillar. Yaml --batch_size 4' command Raise AttributeError Indicates an error. How can I fix it?

Here is my console log and the error returned：
2022-03-15 11:08:47,537   INFO  **********************Start logging**********************
2022-03-15 11:08:47,537   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-03-15 11:08:47,537   INFO  cfg_file         cfgs/tower_models/my_pointpillar.yaml
2022-03-15 11:08:47,537   INFO  batch_size       4
2022-03-15 11:08:47,537   INFO  epochs           80
2022-03-15 11:08:47,537   INFO  workers          8
2022-03-15 11:08:47,537   INFO  extra_tag        default
2022-03-15 11:08:47,537   INFO  ckpt             None
2022-03-15 11:08:47,537   INFO  pretrained_model None
2022-03-15 11:08:47,537   INFO  launcher         none
2022-03-15 11:08:47,537   INFO  tcp_port         18888
2022-03-15 11:08:47,538   INFO  sync_bn          False
2022-03-15 11:08:47,538   INFO  fix_random_seed  False
2022-03-15 11:08:47,538   INFO  ckpt_save_interval 1
2022-03-15 11:08:47,538   INFO  local_rank       0
2022-03-15 11:08:47,538   INFO  max_ckpt_save_num 30
2022-03-15 11:08:47,538   INFO  merge_all_iters_to_one_epoch False
2022-03-15 11:08:47,538   INFO  set_cfgs         None
2022-03-15 11:08:47,538   INFO  max_waiting_mins 0
2022-03-15 11:08:47,538   INFO  start_epoch      0
2022-03-15 11:08:47,538   INFO  save_to_file     False
2022-03-15 11:08:47,538   INFO  cfg.ROOT_DIR: /data1/home/liyao/qfli/fjt/OpenPCDet
2022-03-15 11:08:47,538   INFO  cfg.LOCAL_RANK: 0
2022-03-15 11:08:47,538   INFO  cfg.CLASS_NAMES: ['Tower']
2022-03-15 11:08:47,538   INFO  
cfg.DATA_CONFIG = edict()
2022-03-15 11:08:47,538   INFO  cfg.DATA_CONFIG.DATASET: TowerDataset
2022-03-15 11:08:47,538   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/tower
2022-03-15 11:08:47,538   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -39.68, -3, 69.12, 39.68, 1]
2022-03-15 11:08:47,538   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-03-15 11:08:47,538   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2022-03-15 11:08:47,538   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-03-15 11:08:47,538   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2022-03-15 11:08:47,538   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['tower_infos_train.pkl']
2022-03-15 11:08:47,538   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['tower_infos_val.pkl']
2022-03-15 11:08:47,538   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2022-03-15 11:08:47,539   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2022-03-15 11:08:47,539   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-03-15 11:08:47,539   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2022-03-15 11:08:47,539   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2022-03-15 11:08:47,539   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-03-15 11:08:47,539   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2022-03-15 11:08:47,539   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2022-03-15 11:08:47,539   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2022-03-15 11:08:47,539   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.16, 0.16, 4], 'MAX_POINTS_PER_VOXEL': 32, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2022-03-15 11:08:47,539   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/tower_dataset.yaml
2022-03-15 11:08:47,539   INFO  
cfg.MODEL = edict()
2022-03-15 11:08:47,539   INFO  cfg.MODEL.NAME: PointPillar
2022-03-15 11:08:47,539   INFO  
cfg.MODEL.VFE = edict()
2022-03-15 11:08:47,539   INFO  cfg.MODEL.VFE.NAME: PillarVFE
2022-03-15 11:08:47,539   INFO  cfg.MODEL.VFE.WITH_DISTANCE: False
2022-03-15 11:08:47,539   INFO  cfg.MODEL.VFE.USE_ABSLOTE_XYZ: True
2022-03-15 11:08:47,539   INFO  cfg.MODEL.VFE.USE_NORM: True
2022-03-15 11:08:47,539   INFO  cfg.MODEL.VFE.NUM_FILTERS: [64]
2022-03-15 11:08:47,539   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2022-03-15 11:08:47,539   INFO  cfg.MODEL.MAP_TO_BEV.NAME: PointPillarScatter
2022-03-15 11:08:47,539   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 64
2022-03-15 11:08:47,539   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2022-03-15 11:08:47,539   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
[... 55 lines cut ...]
2022-03-15 11:08:47,541   INFO  cfg.OPTIMIZATION.LR_DECAY: 0.1
2022-03-15 11:08:47,541   INFO  cfg.OPTIMIZATION.LR_CLIP: 1e-07
2022-03-15 11:08:47,541   INFO  cfg.OPTIMIZATION.LR_WARMUP: False
2022-03-15 11:08:47,542   INFO  cfg.OPTIMIZATION.WARMUP_EPOCH: 1
2022-03-15 11:08:47,542   INFO  cfg.OPTIMIZATION.GRAD_NORM_CLIP: 10
2022-03-15 11:08:47,542   INFO  cfg.TAG: my_pointpillar
2022-03-15 11:08:47,542   INFO  cfg.EXP_GROUP_PATH: tower_models
Traceback (most recent call last):
  File "my_train.py", line 236, in <module>
    main()
  File "my_train.py", line 132, in main
    total_epochs=args.epochs
  File "../pcdet/datasets/__init__.py", line 91, in build_dataloader
    drop_last=False, sampler=sampler, timeout=0
  File "/data1/home/liyao/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 268, in __init__
    sampler = RandomSampler(dataset, generator=generator)
  File "/data1/home/liyao/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/torch/utils/data/sampler.py", line 101, in __init__
    if not isinstance(self.num_samples, int) or self.num_samples <= 0:
  File "/data1/home/liyao/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/torch/utils/data/sampler.py", line 109, in num_samples
    return len(self.data_source)
  File "../pcdet/datasets/tower/tower_dataset.py", line 361, in __len__
    return len(self.tower_infos)
  File "/data1/home/liyao/anaconda3/envs/OpenPCDet/lib/python3.7/site-packages/torch/utils/data/dataset.py", line 83, in __getattr__
    raise AttributeError
AttributeError

LouisSF, 2022-03-15:
Hi, without seeing your code this is quite hard to debug but your tower dataset instance, which I assume you created a class for, does not seem to have a tower_infos attribute. Maybe it is not initialized in the `__init__` of the class ?

969682997, 2022-03-15:
> Hi, without seeing your code this is quite hard to debug but your tower dataset instance, which I assume you created a class for, does not seem to have a tower_infos attribute. Maybe it is not initialized in the `__init__` of the class ?

————————————————————————————————————————————————
Thank you for your answer. I checked my own code as you guessed. The tower_infos definition is found in the __init__ method of the TowerDataset class.
Here is the TowerDataset class in my tower_dataset.py, hopefully useful.
————————————————————————————————————————————————
class TowerDataset(DatasetTemplate):
    def init(self, dataset_cfg, class_names, training=True, root_path=None, logger=None):
        """
        Args:
            root_path:
            dataset_cfg:
            class_names:
            training:
            logger:
        """
        super().__init__(
            dataset_cfg=dataset_cfg, class_names=class_names, training=training, root_path=root_path, logger=logger
        )
        # 传递参数是 训练集train 还是验证集val
        self.split = self.dataset_cfg.DATA_SPLIT[self.mode]
        # root_path的路径是../data/tower/
        # tower数据集一共三个文件夹“training”和“testing”、“ImageSets”
        # 如果是训练集train，将文件的路径指为训练集training ，否则为测试集testing
        self.root_split_path = self.root_path / ('training' if self.split != 'test' else 'testing')
        # /data/tower/ImageSets/下面一共三个文件：test.txt , train.txt ,val.txt
        # 选择其中的一个文件
        split_dir = self.root_path / 'ImageSets' / (self.split + '.txt')
        # 得到.txt文件下的序列号，组成列表sample_id_list
        self.sample_id_list = [x.strip() for x in open(split_dir).readlines()] if split_dir.exists() else None
        # 创建用于存放tower信息的空列表
        self.tower_infos = []
        # 调用函数，加载tower数据，mode的值为：train 或者  test
        self.include_tower_data(self.mode)

MartinHahner, 2022-03-22:
How does your `self.include_tower_data(self, mode)` method look like?

closed 2022-05-07 by github-actions[bot]