open-mmlab/OpenPCDet #872: How to turn the single frame of the demo into a video to view？

Leozyc-waseda, 2022-03-15:
I have a continuous frame and I want to see the output as a video, can someone please give me some advice?

LouisSF, 2022-03-15:
Hi, you will have to do modifications with Open3D to modify the code, using this [StackOverflow issue](https://stackoverflow.com/questions/62912397/open3d-visualizing-multiple-point-clouds-as-a-video-animation) and the [Open3D page relative to non-blocking visualization](http://www.open3d.org/docs/0.8.0/tutorial/Advanced/non_blocking_visualization.html)

Then, either use an external recording software or search the Open3D docs, I know you can save individual images of point clouds but I'm not sure you can do the same for videos.

Leozyc-waseda, 2022-04-15:
@LouisSF 
Thank you for your reply,
Do you mean to modify the Open3D display code, change data from 1 frame display to video display?

closed 2022-05-30 by github-actions[bot]

maayanYa, 2022-06-26:
@Leozyc-waseda  Hi , did you understand how to render the bounding boxes in each frame ?
The mentioned examples are related only to the point cloud .