open-mmlab/OpenPCDet #873: How to download Nuscenes dataset easily ?

daeunni, 2022-03-15:
Hi, 
I want to download **Nuscence dataset**, 
but I found that the file on the official site(https://www.nuscenes.org/download) was too large. 

If there's anyone who knows how to download this easily, please let me know!

Best,

LouisSF, 2022-03-15:
Hi, what are you downloading exactly on the page? Nuscenes is already divided into chunks to download on the page. You could also download the mini subset of the dataset if you just want to do some manipulations with the data and not train a model.

daeunni, 2022-03-15:
Thx for reply. :) 

Im trying to download **Full dataset (v1.0) Trainval** because i want to make fully trained model. 
btw, I couldn't find a **separate link** to download Full dataset (v1.0) (including train/val NOT mini dataset). 
It just exists together in a formal download official link(https://www.nuscenes.org/download) that collects many kinds of other data.

If there are separate links to download **Full dataset (v1.0) Trainval** , I think I can download it using `wget.` 
Let me know where this is!

daeunni, 2022-03-15:
[image]

i want download this file

LouisSF, 2022-03-15:
Isn't it possible to download the individual file blobs and then remove the data modalities (such as images) that you don't want ?

daeunni, 2022-03-15:
um.. it might be possible. 
btw individual file's memory is tooo big, so i wanna know how to download it **easier**. 

do you know the Linux command that efficiently downloads files on all pages?

closed 2022-03-22 by daeunni

hygxy, 2022-06-28:
> um.. it might be possible. btw individual file's memory is tooo big, so i wanna know how to download it **easier**.
> 
> do you know the Linux command that efficiently downloads files on all pages?

Did you figure out how to download the full v1.0 dataset using the "wget" Linux command?

dthuremella, 2022-08-02:
I'm also looking for the answer to this

daeunni, 2022-08-03:
@dthuremella  you can use "wget" command

li-xl, 2023-12-02:
I just wrote a Python script that makes it easier to download the current dataset. You can give it a try here: [https://github.com/li-xl/nuscenes-download](https://github.com/li-xl/nuscenes-download).

AshDrift98, 2024-03-02:
i want to download the training dataset of nuScenes in colab.

will this work?

!wget https://www.nuscenes.org/data/v1.0-trainval.tgz```

!tar -xzf v1.0-trainval.tgz

rajeshthamarana, 2024-03-13:
is above commands worked

iampujan, 2025-04-20:
I created a trainval blob downloader and test blob downloader using a bash script. It can be helpful. 
This is for nuscenes v1.0 full dataset.
Follow the readme here: https://github.com/iampujan/download_nuscenes_dataset

CrusarAppy, 2025-07-22:
Had you guys downlaoded the nuscenes datasets , through wget, what is the correct url . Help me on this. Thank you!