open-mmlab/OpenPCDet #876: Question about the gt_sampling data augmentaion of 20% waymo dataset.

azhuantou, 2022-03-16:
Hi, Thanks for your great efforts.
When using 20% of Waymo dataset, the code used the complete "gt_database" for "gt_sampling" data augmentation, causing the baseline performance of 20% data may be too high (which means 100% bounding boxes and points enclosed in them for data augmentation).  Is the baseline of 20% waymo data presented in readme use the 100% gt_datase?

tianweiy, 2022-03-17:
> Is the baseline of 20% waymo data presented in readme use the 100% gt_datase?
I think so 

> causing the baseline performance of 20% data may be too high
I would say this is a nice feature, as it makes iteration 5x faster (while the performance gap is much smaller due to the model being able to see all objects through gt_sampling)

azhuantou, 2022-03-17:
@tianweiy Thanks for your reply. I agree that using 20% data is a good solution, but I'm still confused. When we use 20% of the data, shouldn't we only use the gt_database generated from 20% of the data (instead gt_database generate from 100% data) ?

sshaoshuai, 2022-03-20:
Note that 20% training setting is just for conducting some ablation studies, and of course you can also generate gt_database with only 20% data (I think the performance should not drop a lot). 
But for now, in order to avoid generating  gt_database for each setting (e.g., 20%, 10%, 5%), I just keep to use the same gt_database for all settings.

azhuantou, 2022-03-21:
Thank you for your reply.
I will test the impact of gt_database.

closed 2022-03-21 by azhuantou