open-mmlab/OpenPCDet #880: Is training based on pre-training model not supported?

gzchenjiajun, 2022-03-18:
Is training based on pre-training model not supported? The secend, for example, pointpiller.
I get an error when I use a similar command

python train.py --cfg_file   .. /tools/cfgs/kitti_models/pointpillar.yaml     --ckpt         .. /checkpoints/pointpillar_7728.pth

Traceback (most recent call last):
  File "train.py", line 204, in <module>
    main()
  File "train.py", line 130, in main
    it, start_epoch = model.load_params_with_optimizer(args.ckpt, to_cpu=dist_train, optimizer=optimizer, logger=logger)
  File "../pcdet/models/detectors/detector3d_template.py", line 392, in load_params_with_optimizer
    self._load_state_dict(checkpoint['model_state'], strict=True)
  File "../pcdet/models/detectors/detector3d_template.py", line 355, in _load_state_dict
    self.load_state_dict(update_model_state)
  File "/home/psdz/Desktop/tensorrt_env/TensorRT-6.0.1.5/lib/3dpc/lib/python3.6/site-packages/torch/nn/modules/module.py", line 830, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for SECONDNet:
	Missing key(s) in state_dict: "dense_head.conv_cls.weight", "dense_head.conv_cls.bias", "dense_head.conv_box.weight", "dense_head.conv_box.bias", "dense_head.conv_dir_cls.weight", "dense_head.conv_dir_cls.bias".

jihanyang, 2022-03-18:
Use `--pretrained_model` rather than `--ckpt` can suit your demand.

closed 2022-03-18 by jihanyang

gzchenjiajun, 2022-03-19:
2022-03-19 11:59:14,717   INFO  ==> Loading parameters from checkpoint ../checkpoints/pointpillar_7728.pth to GPU
Traceback (most recent call last):
  File "train.py", line 204, in <module>
    main()
  File "train.py", line 127, in main
    model.load_params_from_file(filename=args.pretrained_model, to_cpu=dist_train, logger=logger)
  File "../pcdet/models/detectors/detector3d_template.py", line 374, in load_params_from_file
    state_dict, update_model_state = self._load_state_dict(model_state_disk, strict=True)
  File "../pcdet/models/detectors/detector3d_template.py", line 355, in _load_state_dict
    self.load_state_dict(update_model_state)
  File "/home/psdz/Desktop/tensorrt_env/TensorRT-6.0.1.5/lib/3dpc/lib/python3.6/site-packages/torch/nn/modules/module.py", line 830, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for PointPillar:
    Missing key(s) in state_dict: "dense_head.conv_cls.weight", "dense_head.conv_cls.bias", "dense_head.conv_box.weight", "dense_head.conv_box.bias", "dense_head.conv_dir_cls.weight", "dense_head.conv_dir_cls.bias". 

 python train.py --cfg_file   ../tools/cfgs/kitti_models/pointpillar.yaml     --pretrained_model          ../checkpoints/pointpillar_7728.pth

same error @jihanyang

gzchenjiajun, 2022-03-19:
(/home/psdz/Desktop/tensorrt_env/TensorRT-6.0.1.5/lib/3dpc) psdz@PSDZ:~/Desktop/fgwj_open_pc_det/tools$ python train.py --cfg_file   ../tools/cfgs/kitti_models/second.yaml    --pretrained_model ../checkpoints/second_7862.pth
2022-03-19 12:05:46,025   INFO  **********************Start logging**********************
2022-03-19 12:05:46,025   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-03-19 12:05:46,025   INFO  cfg_file         ../tools/cfgs/kitti_models/second.yaml
2022-03-19 12:05:46,025   INFO  batch_size       4
2022-03-19 12:05:46,025   INFO  epochs           80
2022-03-19 12:05:46,025   INFO  workers          4
2022-03-19 12:05:46,025   INFO  extra_tag        default
2022-03-19 12:05:46,025   INFO  ckpt             None
2022-03-19 12:05:46,025   INFO  pretrained_model ../checkpoints/second_7862.pth
2022-03-19 12:05:46,025   INFO  launcher         none
2022-03-19 12:05:46,025   INFO  tcp_port         18888
2022-03-19 12:05:46,025   INFO  sync_bn          False
2022-03-19 12:05:46,025   INFO  fix_random_seed  False
2022-03-19 12:05:46,025   INFO  ckpt_save_interval 1
2022-03-19 12:05:46,025   INFO  local_rank       0
2022-03-19 12:05:46,025   INFO  max_ckpt_save_num 30
2022-03-19 12:05:46,025   INFO  merge_all_iters_to_one_epoch False
2022-03-19 12:05:46,025   INFO  set_cfgs         None
2022-03-19 12:05:46,025   INFO  max_waiting_mins 0
2022-03-19 12:05:46,025   INFO  start_epoch      0
2022-03-19 12:05:46,025   INFO  num_epochs_to_eval 0
2022-03-19 12:05:46,025   INFO  save_to_file     False
2022-03-19 12:05:46,025   INFO  cfg.ROOT_DIR: /home/psdz/Desktop/fgwj_open_pc_det
2022-03-19 12:05:46,026   INFO  cfg.LOCAL_RANK: 0
2022-03-19 12:05:46,026   INFO  cfg.CLASS_NAMES: ['car', 'bus', 'dog', 'pedestrian', 'bimo', 'truck']
2022-03-19 12:05:46,026   INFO  
cfg.DATA_CONFIG = edict()
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.DATA_PATH: /media/psdz/SSD_1T/kitti_own/
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -50, -3, 80, 50, 4]
2022-03-19 12:05:46,026   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-03-19 12:05:46,026   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2022-03-19 12:05:46,026   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': [], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:30', 'bus:30', 'dog:30', 'pedestrian:30', 'bimo:30', 'truck:30'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2022-03-19 12:05:46,026   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2022-03-19 12:05:46,026   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2022-03-19 12:05:46,027   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.175], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 14445, 'test': 14445}}]
2022-03-19 12:05:46,027   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2022-03-19 12:05:46,027   INFO  
cfg.MODEL = edict()
2022-03-19 12:05:46,027   INFO  cfg.MODEL.NAME: SECONDNet
2022-03-19 12:05:46,027   INFO  
cfg.MODEL.VFE = edict()
2022-03-19 12:05:46,027   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2022-03-19 12:05:46,027   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2022-03-19 12:05:46,027   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2022-03-19 12:05:46,027   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2022-03-19 12:05:46,027   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2022-03-19 12:05:46,027   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2022-03-19 12:05:46,027   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2022-03-19 12:05:46,027   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2022-03-19 12:05:46,027   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5]
2022-03-19 12:05:46,027   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2]
[... 59 lines cut ...]
2022-03-19 12:05:46,029   INFO  cfg.EXP_GROUP_PATH: tools/cfgs/kitti_models
2022-03-19 12:05:49,953   INFO  Database filter by difficulty car: 272498 => 272498
[... cut ...]
    self.load_state_dict(update_model_state)
  File "/home/psdz/Desktop/tensorrt_env/TensorRT-6.0.1.5/lib/3dpc/lib/python3.6/site-packages/torch/nn/modules/module.py", line 830, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for SECONDNet:
    Missing key(s) in state_dict: "dense_head.conv_cls.weight", "dense_head.conv_cls.bias", "dense_head.conv_box.weight", "dense_head.conv_box.bias", "dense_head.conv_dir_cls.weight", "dense_head.conv_dir_cls.bias". 

how solve

gzchenjiajun, 2022-03-19:
Before that I have performed the python - m pcdet datasets. Kitti. Kitti_dataset create_kitti_infos tools/CFGS dataset_configs/kitti_dataset yaml

I trained with my own data set

gzchenjiajun, 2022-03-19:
its ok
Start training

gzchenjiajun, 2022-03-19:
2022-03-19 12:32:14,385   INFO  **********************Start training tools/cfgs/kitti_models/pointpillar(default)**********************
epochs:   0%|                                                                                                                                                                                                                                               | 0/80 [00:31<?, ?it/s]
Traceback (most recent call last):                                                                                                                                                                                                                        | 0/2064 [00:00<?, ?it/s]
  File "train.py", line 204, in <module>
    main()
  File "train.py", line 171, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/psdz/Desktop/fgwj_open_pc_det/tools/train_utils/train_utils.py", line 118, in train_model
    dataloader_iter=dataloader_iter
  File "/home/psdz/Desktop/fgwj_open_pc_det/tools/train_utils/train_utils.py", line 47, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "../pcdet/models/__init__.py", line 42, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/psdz/Desktop/tensorrt_env/TensorRT-6.0.1.5/lib/3dpc/lib/python3.6/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "../pcdet/models/detectors/pointpillar.py", line 14, in forward
    loss, tb_dict, disp_dict = self.get_training_loss()
  File "../pcdet/models/detectors/pointpillar.py", line 27, in get_training_loss
    loss_rpn, tb_dict = self.dense_head.get_loss()
  File "../pcdet/models/dense_heads/anchor_head_template.py", line 217, in get_loss
    cls_loss, tb_dict = self.get_cls_layer_loss()
  File "../pcdet/models/dense_heads/anchor_head_template.py", line 128, in get_cls_layer_loss
    cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights)  # [N, M]
  File "/home/psdz/Desktop/tensorrt_env/TensorRT-6.0.1.5/lib/3dpc/lib/python3.6/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "../pcdet/utils/loss_utils.py", line 59, in forward
    pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid
RuntimeError: The size of tensor a (40176) must match the size of tensor b (642816) at non-singleton dimension 1

new error....

zjr-bit, 2022-03-24:
You can solve the new error by configure the point range along x,y-axis / voxel size anlong x,y-axis is the multiple of 16

huangqzj, 2022-04-21:
help @zjr-bit , i encounter this error:
Traceback (most recent call last):
  File "train.py", line 203, in <module>
    main()
  File "train.py", line 158, in main
    train_model(
  File "/root/OpenPCDet/CenterPoint-Merge/tools/train_utils/train_utils.py", line 87, in train_model
    accumulated_iter = train_one_epoch(
  File "/root/OpenPCDet/CenterPoint-Merge/tools/train_utils/train_utils.py", line 39, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/__init__.py", line 30, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/detectors/pointpillar.py", line 15, in forward
    loss, tb_dict, disp_dict = self.get_training_loss()
  File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/detectors/pointpillar.py", line 28, in get_training_loss
    loss_rpn, tb_dict = self.dense_head.get_loss()
  File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/dense_heads/anchor_head_template.py", line 217, in get_loss
    cls_loss, tb_dict = self.get_cls_layer_loss()
  File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/dense_heads/anchor_head_template.py", line 128, in get_cls_layer_loss
    cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights)  # [N, M]
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/OpenPCDet/CenterPoint-Merge/pcdet/utils/loss_utils.py", line 60, in forward
    pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid
RuntimeError: The size of tensor a (34656) must match the size of tensor b (554496) at non-singleton dimension 1.
My configure is :
POINT_CLOUD_RANGE: [-14.30, -18.60, 0, 82.98, 78.68, 4] 
VOXEL_SIZE: [0.16, 0.16, 4]
MAX_POINTS_PER_VOXEL: 32
MAX_NUMBER_OF_VOXELS: {
            'train': 216000,
            'test': 40000
          }
the point range along x,y-axis / voxel size  is the multiple of 16.but still this error.

hanjiangrui, 2022-05-19:
> help @zjr-bit , i encounter this error: Traceback (most recent call last): File "train.py", line 203, in main() File "train.py", line 158, in main train_model( File "/root/OpenPCDet/CenterPoint-Merge/tools/train_utils/train_utils.py", line 87, in train_model accumulated_iter = train_one_epoch( File "/root/OpenPCDet/CenterPoint-Merge/tools/train_utils/train_utils.py", line 39, in train_one_epoch loss, tb_dict, disp_dict = model_func(model, batch) File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/**init**.py", line 30, in model_func ret_dict, tb_dict, disp_dict = model(batch_dict) File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl result = self.forward(*input, **kwargs) File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/detectors/pointpillar.py", line 15, in forward loss, tb_dict, disp_dict = self.get_training_loss() File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/detectors/pointpillar.py", line 28, in get_training_loss loss_rpn, tb_dict = self.dense_head.get_loss() File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/dense_heads/anchor_head_template.py", line 217, in get_loss cls_loss, tb_dict = self.get_cls_layer_loss() File "/root/OpenPCDet/CenterPoint-Merge/pcdet/models/dense_heads/anchor_head_template.py", line 128, in get_cls_layer_loss cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights) # [N, M] File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 727, in _call_impl result = self.forward(*input, **kwargs) File "/root/OpenPCDet/CenterPoint-Merge/pcdet/utils/loss_utils.py", line 60, in forward pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid RuntimeError: The size of tensor a (34656) must match the size of tensor b (554496) at non-singleton dimension 1. My configure is : POINT_CLOUD_RANGE: [-14.30, -18.60, 0, 82.98, 78.68, 4] VOXEL_SIZE: [0.16, 0.16, 4] MAX_POINTS_PER_VOXEL: 32 MAX_NUMBER_OF_VOXELS: { 'train': 216000, 'test': 40000 } the point range along x,y-axis / voxel size is the multiple of 16.but still this error.

Have you solved the problem？ i meet a same one

Christina-Soda, 2022-07-24:
> its ok
> Start training

Could you please tell me how to start training from above error? Thanks!