open-mmlab/OpenPCDet #883: Why is there a "GroupingOperation"?

demmerichs, 2022-03-18:
My question is about a specific function defined [here](https://github.com/open-mmlab/OpenPCDet/blob/a991105c1b11bbb4dc6374eb33f0a4c51049c88d/pcdet/ops/pointnet2/pointnet2_batch/pointnet2_utils.py#L156-L197)!

It seems to me, that this function could easily be replaced by torch.gather in combination with some view manipulation.

So I was wondering if it has some optimization done, that goes beyond torch.gather. I am trying to learn how to better optimize my networks.

I propose and tested it on a small example, that the following line would do the same as the aforementioned GroupingOperation:
```python
out = grouping_operation(features, idx)
out = torch.gather(features[..., None].expand(-1, -1, -1, idx.size(2)), 2, idx.to(torch.long)[:, None].expand(-1, features.size(1), -1, -1))
```
Note, that PyTorch documentation specifies, that torch.expand just changes the view by modifying the stride parameter and not allocating new memory.

Any insights on this design decision would be greatly appreciated :)

sshaoshuai, 2022-03-20:
Yes, you are right - This operation can be replaced with torch.gather or just simple indexing.
It is still here since this is a legacy code and also sometimes it runs faster than the default indexing of pytorch.

demmerichs, 2022-03-21:
Ok, thanks. But that means nowadays there is no need for this grouping, as torch.gather should perform similarly fast.

closed 2022-03-21 by demmerichs