open-mmlab/OpenPCDet #884: AP is 0

nicholaswhf, 2022-03-18:
I got 0 AP when evaluate the pretrained SECOND model.
2022-03-19 00:05:08,592   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:90.7803, 89.8999, 89.0420
bev  AP:90.0097, 87.9166, 86.4514
3d   AP:88.6137, 78.6186, 77.2174
aos  AP:90.76, 89.77, 88.82
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:95.5933, 94.1555, 91.7341
bev  AP:92.3844, 88.5242, 87.6145
3d   AP:90.5238, 81.5823, 78.5787
aos  AP:95.56, 93.99, 91.48
Car AP@0.70, 0.50, 0.50:
bbox AP:90.7803, 89.8999, 89.0420
bev  AP:90.7940, 90.1441, 89.5180
3d   AP:90.7940, 90.0886, 89.4014
aos  AP:90.76, 89.77, 88.82
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:95.5933, 94.1555, 91.7341
bev  AP:95.6398, 94.8139, 94.2144
3d   AP:95.6274, 94.7115, 94.0238
aos  AP:95.56, 93.99, 91.48
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00

laiyingxin2, 2022-03-23:
What changes have you made so that the accuracy is not 0

closed 2022-05-08 by github-actions[bot]

cyb1219257828, 2024-03-07:
> I got 0 AP when evaluate the pretrained SECOND model. 2022-03-19 00:05:08,592 INFO Car AP@0.70, 0.70, 0.70: bbox AP:90.7803, 89.8999, 89.0420 bev AP:90.0097, 87.9166, 86.4514 3d AP:88.6137, 78.6186, [...]

Did you solve it? I also encountered the same problem.

thanhpt93, 2025-01-03:
@nicholaswhf  Did you solve it? I also encountered the same problem.