open-mmlab/OpenPCDet #887: error when loss computation

zjr-bit, 2022-03-19:
hello everyone! I 'm trying train PointPillars using my own dataset. When I train the network, the error come out:

Traceback (most recent call last):
  File "train.py", line 202, in <module>
    main()
  File "train.py", line 154, in main
    train_model(
  File "/home/zjr/OpenPCDet/tools/train_utils/train_utils.py", line 111, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/zjr/OpenPCDet/tools/train_utils/train_utils.py", line 47, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/__init__.py", line 54, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/zjr/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/detectors/pointpillar.py", line 14, in forward
    loss, tb_dict, disp_dict = self.get_training_loss()
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/detectors/pointpillar.py", line 27, in get_training_loss
    loss_rpn, tb_dict = self.dense_head.get_loss()
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/dense_heads/anchor_head_template.py", line 217, in get_loss
    cls_loss, tb_dict = self.get_cls_layer_loss()
  File "/home/zjr/OpenPCDet/tools/../pcdet/models/dense_heads/anchor_head_template.py", line 128, in get_cls_layer_loss
    cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights)  # [N, M]
  File "/home/zjr/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/zjr/OpenPCDet/tools/../pcdet/utils/loss_utils.py", line 59, in forward
    pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid
RuntimeError: The size of tensor a (16920) must match the size of tensor b (17280) at non-singleton dimension 1
train:   0%|                                              | 0/3 [00:00<?, ?it/s]

I try to solve it, but in vain. Please give me some help! Thanks a lot!

zjr-bit, 2022-03-19:
Another question is :
I set this in the dataset.yaml

POINT_FEATURE_ENCODING: { # 点特征编码
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity'],
}

then when training the network, I found that in models/__init__.py load_data_to_gpu(batch_dict), the dimension of 'points' is (114588, 5). I don't know why the number is 5.

Please tell me! Thanks a lot!

zjr-bit, 2022-03-19:
> Another question is : I set this in the dataset.yaml
>
> POINT_FEATURE_ENCODING: { # 点特征编码 encoding_type: absolute_coordinates_encoding, used_feature_list: ['x', 'y', 'z', 'intensity'], [...]

Now I know: it adds the batch_size to the points

closed 2022-03-20 by zjr-bit

KyleYueye, 2022-07-03:
> hello everyone! I 'm trying to train PointPillars using my own dataset. When I am training the network, the error comes out:
> 
> Traceback (most recent call last): File "train.py", line 202, in main() File "train.py", line 154, in main train_model( File "/home/zjr/OpenPCDet/tools/train_utils/train_utils.py", line 111, in train_model accumulated_iter = train_one_epoch( File "/home/zjr/OpenPCDet/tools/train_utils/train_utils.py", line 47, in train_one_epoch loss, tb_dict, disp_dict = model_func(model, batch) File "/home/zjr/OpenPCDet/tools/../pcdet/models/**init**.py", line 54, in model_func ret_dict, tb_dict, disp_dict = model(batch_dict) File "/home/zjr/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 550, in **call** result = self.forward(*input, **kwargs) File "/home/zjr/OpenPCDet/tools/../pcdet/models/detectors/pointpillar.py", line 14, in forward loss, tb_dict, disp_dict = self.get_training_loss() File "/home/zjr/OpenPCDet/tools/../pcdet/models/detectors/pointpillar.py", line 27, in get_training_loss loss_rpn, tb_dict = self.dense_head.get_loss() File "/home/zjr/OpenPCDet/tools/../pcdet/models/dense_heads/anchor_head_template.py", line 217, in get_loss cls_loss, tb_dict = self.get_cls_layer_loss() File "/home/zjr/OpenPCDet/tools/../pcdet/models/dense_heads/anchor_head_template.py", line 128, in get_cls_layer_loss cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights) # [N, M] File "/home/zjr/anaconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 550, in **call** result = self.forward(*input, **kwargs) File "/home/zjr/OpenPCDet/tools/../pcdet/utils/loss_utils.py", line 59, in forward pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid RuntimeError: The size of tensor a (16920) must match the size of tensor b (17280) at non-singleton dimension 1 train: 0%| | 0/3 [00:00<?, ?it/s]
> 
> I try to solve it, but in vain. Please give me some help! Thanks a lot!

How do you solve it? I have the same question