open-mmlab/OpenPCDet #890: when train I get zeros in evaluation

zjr-bit, 2022-03-21:
Hello everyone!
I am trying to train PointPillars on DAIR-V2X dataset. I have complete the dair_dataset.py similar to kitti_dataset.py and I can train without error. But when finish training, I get all zeros in the module of evaluation. The information is as follows:

> 2022-03-21 16:44:40,422   INFO  **********************Start evaluation dair_models/pointpillar(default)**********************
2022-03-21 16:44:40,422   INFO  Loading DAIR dataset
2022-03-21 16:44:40,423   INFO  Total samples for DAIR dataset: 5
2022-03-21 16:44:40,425   INFO  ==> Loading parameters from checkpoint /home/zjr/OpenPCDet/output/dair_models/pointpillar/default/ckpt/checkpoint_epoch_1.pth to GPU
2022-03-21 16:44:40,460   INFO  ==> Checkpoint trained from version: pcdet+0.5.2+a35f429
2022-03-21 16:44:40,466   INFO  ==> Done (loaded 127/127)
2022-03-21 16:44:40,468   INFO  *************** EPOCH 1 EVALUATION *****************
eval: 100%|██████████| 5/5 [00:00<00:00, 21.15it/s, recall_0.3=(0, 0) / 146]
2022-03-21 16:44:40,705   INFO  *************** Performance of EPOCH 1 *****************
2022-03-21 16:44:40,706   INFO  Generate label finished(sec_per_example: 0.0474 second).
2022-03-21 16:44:40,706   INFO  recall_roi_0.3: 0.000000
2022-03-21 16:44:40,706   INFO  recall_rcnn_0.3: 0.000000
2022-03-21 16:44:40,706   INFO  recall_roi_0.5: 0.000000
2022-03-21 16:44:40,706   INFO  recall_rcnn_0.5: 0.000000
2022-03-21 16:44:40,706   INFO  recall_roi_0.7: 0.000000
2022-03-21 16:44:40,706   INFO  recall_rcnn_0.7: 0.000000
2022-03-21 16:44:40,706   INFO  Average predicted number of objects(5 samples): 0.000
2022-03-21 16:44:40,707   INFO  None
2022-03-21 16:44:40,707   INFO  Result is save to /home/zjr/OpenPCDet/output/dair_models/pointpillar/default/eval/eval_with_train/epoch_1/val
2022-03-21 16:44:40,707   INFO  ****************Evaluation done.*****************
2022-03-21 16:44:40,708   INFO  Epoch 1 has been evaluated
2022-03-21 16:45:10,739   INFO  **********************End evaluation dair_models/pointpillar(default)**********************

By the way, I add print() in the pcdet/utils/loss_utils.py forward() function to show the value of target. And I found target is a tensor filled with all zeros.  So I am wondering whether this situation is right.

>         print('input: ', input.shape)
>         print('target: ', target.shape)
>         print('target: ', target)
>         print('weights: ', weights.shape)

The output is:
> epochs:   0%|          | 0/1 [00:00<?, ?it/s, loss=3.92, lr=0.00225, d_time=0.00(0.02), f_time=0.03(0.04), b_time=0.06(0.09)]input:  torch.Size([1, 38400, 3])
target:  torch.Size([1, 38400, 3])
target:  tensor([[[0., 0., 0.],
         [0., 0., 0.],
         [0., 0., 0.],
         ...,
         [0., 0., 0.],
         [0., 0., 0.],
         [0., 0., 0.]]], device='cuda:0')
weights:  torch.Size([1, 38400])

@sshaoshuai ， please give me some advice. Thank you very much!!!

zjr-bit, 2022-03-21:
The dair_dataset.py is as follow:

`
import copy
import json
import pickle
import os
import numpy as np
import pandas
import random
from pcdet.ops.roiaware_pool3d import roiaware_pool3d_utils
from pcdet.utils import box_utils, calibration_kitti, common_utils, object3d_kitti
from pcdet.datasets.dataset import DatasetTemplate
from pathlib import Path

class DairDataset(DatasetTemplate):
    def __init__(self, dataset_cfg, class_names, training=True, root_path=None, logger=None):
        """
        Args:
            root_path:
            dataset_cfg:
            class_names:
            training:
            logger:
        """
        super().__init__(
            dataset_cfg=dataset_cfg, class_names=class_names, training=training, root_path=root_path, logger=logger
        )

        # 初始化一个列表
        self.dair_infos = []
        # 用于存放文件路径的列表
        self.files_list_pcd = []
        self.files_list_label = []
        self.files_list_label_train = []
        self.files_list_label_val = []
        self.files_list_pcd_train = []
        self.files_list_pcd_val = []
        self.train_ratio_of_all_labels = self.dataset_cfg.TRAIN_RATIO_OF_ALL_LABELS
        self.split = self.dataset_cfg.DATA_SPLIT[self.mode]

        self.include_dair_data(self.mode)

    def include_dair_data(self, mode):
        if self.logger is not None:
            self.logger.info('Loading DAIR dataset')
        # 新建列表，存储信息
        dair_infos = []
        '''
          INFO_PATH:{
              'train':[dair_infos_train.pkl],
              'test':[dair_infos_val.pkl],}
          '''

        for info_path in self.dataset_cfg.INFO_PATH[mode]:
            info_path = self.root_path / info_path
            if not Path(info_path).exists():
                continue
            with open(info_path, 'rb') as f:
                infos = pickle.load(f)
                dair_infos.extend(infos)

        self.dair_infos.extend(dair_infos)

        if self.logger is not None:
            self.logger.info('Total samples for DAIR dataset: %d' % (len(dair_infos)))

    # 根据数据地址的路径，获取路径下 文件夹的名字列表
    def get_folder_list(self, root_path):
        folder_list = []
[... 563 lines cut ...]

    pass

if __name__ == '__main__':
    import sys

    if sys.argv.__len__() > 1 and sys.argv[1] == 'create_dair_infos':
        import yaml
        from pathlib import Path
        from easydict import EasyDict

        # 读取cfg文件并转为字典
        dataset_cfg = EasyDict(yaml.safe_load(open(sys.argv[2])))
        # ROOT_DIR是得到当前项目的根目录：/home/zjr/OpenPCDet
        ROOT_DIR = (Path(__file__).resolve().parent / '../../../').resolve()
        print(ROOT_DIR)
        print(ROOT_DIR / 'data' / 'dair')
        create_dair_infos(
            dataset_cfg=dataset_cfg,
            class_names=['Car', 'Van', 'Bus'],
            # /home/zjr/OpenPCDet/data/dair
            data_path=ROOT_DIR / 'data' / 'dair',
            save_path=ROOT_DIR / 'data' / 'dair'
        )
`

closed 2022-03-23 by zjr-bit

laiyingxin2, 2022-03-23:
What changes have you made so that the accuracy is not 0

BigPig117, 2022-05-15:
hi, can you use the data set of Dair for training, except dair_ dataset. py, does need to change other files?   I'm a rookie and I'm also trying to train with Dair.  Hope to get your advice