open-mmlab/OpenPCDet #892: ‘Not updated weight’ when I only test Car class

csj777, 2022-03-22:
I try to test only Car class, but  i receive this message:

2022-03-22 15:28:23,185   INFO  ==> Loading parameters from checkpoint ../pv_rcnn_8369.pth to GPU
2022-03-22 15:28:23,272   INFO  Not updated weight dense_head.conv_cls.weight: torch.Size([6, 512, 1, 1])
2022-03-22 15:28:23,273   INFO  Not updated weight dense_head.conv_cls.bias: torch.Size([6])
2022-03-22 15:28:23,273   INFO  ==> Done (loaded 365/367)

and  the Car AP is zero.
like this:
2022-03-22 15:34:00,050   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:1.5152, 1.5152, 1.5152
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00

So now I don't know how to solve it

daeunni, 2022-03-25:
Hi, 
did you trained your own model? 
and except **CAR class** can you evaluate the other classes?

csj777, 2022-03-25:
Yes, you're right! I should train my own model and then test it. That's work! Thank you!

closed 2022-03-26 by csj777