open-mmlab/OpenPCDet #895: what does "anchor_bottom_heights" mean?

slayerlpj, 2022-03-23:
what does "anchor_bottom_heights" mean?
How should I set this value？
for example:
   'class_name': 'Car',
                'anchor_sizes': [[3.9, 1.6, 1.56]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-1.78],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.6,
                'unmatched_threshold': 0.45
Why anchor_bottom_heights is -1.78?

MartinHahner, 2022-03-23:
Because KITTI's LiDAR sensor is located 1.78m above the ground. See [here](http://www.cvlibs.net/datasets/kitti/setup.php).

[image]

slayerlpj, 2022-03-24:
> Because KITTI's LiDAR sensor is located 1.78m above the ground. See [here](http://www.cvlibs.net/datasets/kitti/setup.php).
> 
> [image]

Thank you for your reply！！！

  'class_name': 'Pedestrian',
                'anchor_sizes': [[0.8, 0.6, 1.73]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-0.6],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
'class_name': 'Cyclist',
                'anchor_sizes': [[1.76, 0.6, 1.73]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-0.6],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35

Why pedestrian  and cyclist anchor_bottom_heights is -0.6?

MartinHahner, 2022-03-24:
> Why pedestrian and cyclist anchor_bottom_heights is -0.6?

Good question, no idea. 
Maybe my initial answer was wrong, perhaps try to search in the code where `anchor_bottom_heights` is used.

closed 2022-05-08 by github-actions[bot]

jinhuilin97, 2022-10-06:
There is a question.
if I got the pre-train model by Waymo datasets, and I want to fine-tune in Kitti or other datasets,
what should I do with these part?