open-mmlab/OpenPCDet #896: How to visualize the detection results on the nuscenes dataset

gkd2020, 2022-03-23:
hi，

Thanks for your great work!
I want to visualize detection results.  Specifically, I want to save a lot of visualization results(3d pointcloud image or camera image) and select some of them. But I do not know how to do.
After I run `test.py`,  two result files are generated, `result.pkl` and `results_nusc.json` .
What is the differences between `result.pkl` and `results_nusc.json`.
I guess `results_nusc.json `   uses the nuscenes official format ？Can I export the `results_nusc.json` and save the visual results in batches using the nuscenes-devkit ?

Can you give me some suggestions on how to save the visual results of nuscenes in batches. Many thanks!

MartinHahner, 2022-03-23:
You can visualize `result.pkl` via drag&drop with this [point cloud viewer](https://github.com/MartinHahner/LiDAR_fog_sim/blob/1b303922bf524919d56eaea3406442a067e25d9a/pointcloud_viewer.py) and some code changes on your end.
Eventually, I am trying to create a pull request to OpenPCDet with this point cloud viewer, but did not have the time to do so yet.

gkd2020, 2022-03-25:
> You can visualize `result.pkl` via drag&drop with this [point cloud viewer](https://github.com/MartinHahner/LiDAR_fog_sim/blob/1b303922bf524919d56eaea3406442a067e25d9a/pointcloud_viewer.py) and some [...]

Hi, @MartinHahner 
Thanks for your work,  `pointcloud_viewer.py` can display point cloud files very conveniently.
But, I can not load `result.pickle`  of nuscenes. I followed your prompt and dragged the result.pkl file into the window.

```
Traceback (most recent call last):
  File "/home/zzy/PycharmProjects/LiDAR_fog_sim/pointcloud_viewer.py", line 1872, in dropEvent
    self.create_file_list(Path(self.droppedFilename).parent, self.droppedFilename)
  File "/home/zzy/PycharmProjects/LiDAR_fog_sim/pointcloud_viewer.py", line 1258, in create_file_list
    self.show_pcdet_dict(self.file_list[self.index])
  File "/home/zzy/PycharmProjects/LiDAR_fog_sim/pointcloud_viewer.py", line 1327, in show_pcdet_dict
    pc = pcdet_dict['points']
TypeError: list indices must be integers or slices, not str

```
Then，I tried to operate directly on the GUI

[image]

```
Traceback (most recent call last):
  File "/home/zzy/PycharmProjects/LiDAR_fog_sim/pointcloud_viewer.py", line 979, in load_nuscenes
    self.file_list.append(NUSCENES + file)
TypeError: unsupported operand type(s) for +: 'PosixPath' and 'str'
```
I can't use this tool successfully. Can you tell me how to use it in details  
Many thanks！

MartinHahner, 2022-03-25:
> I can't use this tool successfully. Can you tell me how to use it in detail?

I am sorry, but you have to debug the code yourself and then adjust it to your needs.

closed 2022-03-26 by gkd2020