open-mmlab/OpenPCDet #901: I have encountered the following problems. How can I solve them

libingDY, 2022-03-24:
Traceback (most recent call last):
  File "train.py", line 6, in <module>
    from test import repeat_eval_ckpt
  File "/home/libing/newstorage/3Dobjectdetector/HVPR/tools/test.py", line 13, in <module>
    from eval_utils import eval_utils
  File "/home/libing/newstorage/3Dobjectdetector/HVPR/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "/home/libing/newstorage/3Dobjectdetector/HVPR/pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/home/libing/newstorage/3Dobjectdetector/HVPR/pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/home/libing/newstorage/3Dobjectdetector/HVPR/pcdet/models/detectors/detector3d_template.py", line 7, in <module>
    from .. import backbones_2d, backbones_3d, dense_heads, roi_heads, transformer
  File "/home/libing/newstorage/3Dobjectdetector/HVPR/pcdet/models/backbones_3d/__init__.py", line 1, in <module>
    from .pointnet2_backbone import PointNet2Backbone, PointNet2MSG, PointNet2MSG_NOFP
  File "/home/libing/newstorage/3Dobjectdetector/HVPR/pcdet/models/backbones_3d/pointnet2_backbone.py", line 5, in <module>
    from ...ops.pointnet2.pointnet2_stack import pointnet2_modules as pointnet2_modules_stack
  File "/home/libing/newstorage/3Dobjectdetector/HVPR/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_modules.py", line 7, in <module>
    from . import pointnet2_utils
  File "/home/libing/newstorage/3Dobjectdetector/HVPR/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_utils.py", line 5, in <module>
    from . import pointnet2_stack_cuda as pointnet2
ImportError: /home/libing/newstorage/3Dobjectdetector/HVPR/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_stack_cuda.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _Z25voxel_query_wrapper_stackiiiiifiiiN2at6TensorES0_S0_S0_S0_

6master6, 2022-03-25:
pytorch version is too high, suggest pytorch==1.8.1

libingDY, 2022-03-25:
My environment is as follows：
cuda=10.1
pytorch=1.4.0
So I don't think it's the reason why the pytorch version is too high.

6master6, 2022-03-31:
pytorch version not match pcdet code.

https://blog.csdn.net/weixin_40378209/article/details/123702025

closed 2022-05-16 by github-actions[bot]

20181102931, 2024-12-13:
请问这个问题最后是如何解决的？我遇到了同样的问题？