open-mmlab/OpenPCDet #907: Potential bug in grid_size in DataProcessor

Rick0514, 2022-03-25:
### description
Recently, I apply PointPillars to Kitti dataset and bumped into **Index out of range** problem. I use the point clouds of kitti is fine. However if I transform the point clouds 15 meter forward along the x axis, the error occurs. Partial error message is listed below. Therefore, I reproduced the bug by saving the first point cloud which causes the error(If u want the point cloud file, feel free to contact me via email).

[image]

### env

- python 3.8
- pytorch 1.8
- newest version of pcdet
- spconv-cu113
- cuda 11.2
- RTX 3060Ti
- PointPillars
- POINT_CLOUD_RANGE: [-10, -39.68, -3, 40, 39.68, 1]
- VOXEL_SIZE: [0.16, 0.16, 4]

### Debugging
I shifted to CPU mode to inferencing hoping to get more information and found the coordinates where features of pillars need to be turn back to original voxel to form pseudo-image are out of range. Then I found the coordinates are generated by ```spconv```. As we can see the [code](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/processor/data_processor.py):
```
if self.voxel_generator is None:
    self.voxel_generator = VoxelGeneratorWrapper(
        vsize_xyz=config.VOXEL_SIZE,
        coors_range_xyz=self.point_cloud_range,
        num_point_features=self.num_point_features,
        max_num_points_per_voxel=config.MAX_POINTS_PER_VOXEL,
        max_num_voxels=config.MAX_NUMBER_OF_VOXELS[self.mode],
    )

points = data_dict['points']
voxel_output = self.voxel_generator.generate(points)
voxels, coordinates, num_points = voxel_output

if not data_dict['use_lead_xyz']:
    voxels = voxels[..., 3:]  # remove xyz in voxels(N, 3)

data_dict['voxels'] = voxels
data_dict['voxel_coords'] = coordinates
data_dict['voxel_num_points'] = num_points
```
 
We simply do the math: POINT_CLOUD_RANGE: [-10, -39.68, -3, 40, 39.68, 1] VOXEL_SIZE: [0.16, 0.16, 4]
we get grid_size: x = 50 / 0.16 = 312.5 y = 39.68 * 2 / 0.16 = 496.0
Here is the **Problem!!**

From the pcdet's code, the grid size is calculated:
```
grid_size = (self.point_cloud_range[3:6] - self.point_cloud_range[0:3]) / np.array(config.VOXEL_SIZE)
self.grid_size = np.round(grid_size).astype(np.int64)
self.voxel_size = config.VOXEL_SIZE
``` 

Googling it, np.round returns to the closest even number, which means **x = 312**

But the coordinates are generated by spconv, the grid size it calculates is: [code](https://github.com/traveller59/spconv/blob/fe4a2e61965d719fe2115a554b601d580b536d4c/spconv/csrc/sparse/pointops.py)

```
code.raw(f"""
int64_t prod = 1;
for (size_t i = 0; i < {self.ndim}; ++i) {{
    grid_size[i] =
        std::round((coors_range[{self.ndim} + i] - coors_range[i]) / vsize[i]);
}}
```

Googling it, std::round returns x.5 --> x+1 which means **x = 313**. Therefore it has a low probability to generate 312 index which is beyond the grid size in pcdet.

### Naive solution

add some code to ensure the coodinates to be valid.

```
valid_indices = np.bitwise_and(coordinates[:, 1] < self.grid_size[1], coordinates[:, 2] < self.grid_size[0])
voxels = voxels[valid_indices, ...]
coordinates = coordinates[valid_indices, :]
num_points = num_points[valid_indices]

data_dict['voxels'] = voxels
data_dict['voxel_coords'] = coordinates
data_dict['voxel_num_points'] = num_points
```
 
Then free of **OutOfRange!!**

jihanyang, 2022-03-25:
Hello, thank you for your feedback. Could you provide a PR to fix this problem?

Rick0514, 2022-03-26:
It's not a big deal. The most effective way to avoid the bug is to make sure the ```POINT_CLOUD_RANGE``` to be divisible by the ```VOXEL_SIZE```. Later I found all config files have already ensured that. Maybe I should not modify the range at the beginning and it's just a kinda newbie bug. The issue serves as a reminder for those who would like to modify some parameters and is going to be close soon.

closed 2022-03-26 by Rick0514