open-mmlab/OpenPCDet #909: ImportError: cannot import name 'roiaware_pool3d_cuda' from 'pcdet.ops.roiaware_pool3d'

suvasis, 2022-03-27:
hi,

I have cuda 11.4, pytorch 10.1 in gpu env. How can I remove this following error for Waymo database?

python -m pcdet.datasets.waymo.waymo_dataset --func create_waymo_infos --cfg_file tools/cfgs/dataset_configs/waymo_dataset.yaml
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/pcdet/datasets/__init__.py", line 7, in <module>
    from .dataset import DatasetTemplate
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/pcdet/datasets/dataset.py", line 8, in <module>
    from .augmentor.data_augmentor import DataAugmentor
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 6, in <module>
    from . import augmentor_utils, database_sampler
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/pcdet/datasets/augmentor/augmentor_utils.py", line 5, in <module>
    from ...utils import box_utils
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/pcdet/utils/box_utils.py", line 7, in <module>
    from ..ops.roiaware_pool3d import roiaware_pool3d_utils
  File "/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_utils.py", line 6, in <module>
    from . import roiaware_pool3d_cuda
ImportError: cannot import name 'roiaware_pool3d_cuda' from 'pcdet.ops.roiaware_pool3d' (/home/minasm/suvasis/tools/pvrcnn/OpenPCDet/pcdet/ops/roiaware_pool3d/__init__.py)

closed 2022-05-12 by github-actions[bot]

ManonCortial, 2023-10-26:
I also have this error, I just managed to build MMPCDet with python 3.11.5, pytorch 2.1.0, cuda 12.1, spconv 2.3.6

I can't see any roiaware_pool3d_cuda in the pcdet.ops.roiaware_pool3d foldr in the sources.

NayaBaslan, 2024-06-21:
hello, was anyone able to solve this issue?

kuailexiaohunzi, 2024-09-15:
> hello, was anyone able to solve this issue?

Have you solved it yet? I encountered the same problem

fangweicheng6, 2024-10-18:
> > 你好，有人能解决这个问题吗？
> 
> 你解决了吗？我也遇到了同样的问题

你解决了吗？我也遇到了这个问题

arashfree, 2024-11-12:
you should install setup.py 

see https://github.com/open-mmlab/OpenPCDet/blob/master/docs/INSTALL.md