open-mmlab/OpenPCDet #911: d3_box_overlap_kernel # ONLY support overlap in CAMERA, not lidar.

Anutlet, 2022-03-28:
[https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/kitti/kitti_object_eval_python/eval.py#:~:text=%23%20ONLY%20support%20overlap%20in%20CAMERA%2C%20not%20lider.](url)

I have  lidar 3D detect result to evaluate, and there has no image 2d detect result. when I use it to evaluate,there has some problems.
for test , I provide the same file at the same time as the input,which is  ground truth and detect result , but the result is unnormal.
when I debug the problem, I find that annotate confusing me.I can't understand what is mean.

laiyingxin2, 2022-03-28:
the same problem, there are some problems when evaluating my data set. My AP and other accuracy are 0. At the same time, I set the 2D information to car, 0,0,0,1,2,3,4 , x. Y, Z, Dy, DX, DZ, rotation angle,according to the issue

DJNing, 2022-03-29:
I got the same questions here. And I've also noticed that the dt_annos from CenterPoint is formated in Lidar coordinate instead of camera coordinate. This problem makes all the posted results unreliable.

laiyingxin2, 2022-03-29:
Can anyone help us

DJNing, 2022-03-29:
> pcdet/datasets/kitti/kitti_object_eval_python/eval.py line126
> 
> I have lidar 3D detect result to evaluate, and there has no image 2d detect result. when I use it to evaluate,there has some problems. for test , I provide the same file at the same time as the input,which is ground truth and detect result , but the result is unnormal. when I debug the problem, I find that annotate confusing me.I can't understand what is mean.

For this problem, I think it will give you more or less the same result if your dt_annos and gt_annos are in the same coordinate system, either Lidar or Camera. 

What I am concerned about most is whether they use Lidar coordinate or Camera coordinate in the network during training/inference, especially for functions like ```assigned_targets```. So far, I've checked that they use **Lidar** coordinate in CenterPoint, but I can't find codes that they transform the boxes to **Camera** coordinate.

laiyingxin2, 2022-03-29:
> > pcdet/datasets/kitti/kitti_object_eval_python/eval.py line126
> > I have lidar 3D detect result to evaluate, and there has no image 2d detect result. when I use it to evaluate,there has some problems. for test , I provide the same file at the same time as the input,which is ground truth and detect result , but the result is unnormal. when I debug the problem, I find that annotate confusing me.I can't understand what is mean.
> 
> For this problem, I think it will give you more or less the same result if your dt_annos and gt_annos are in the same coordinate system, either Lidar or Camera. Besides, if you load your gt with function `box_utils.boxes3d_kitti_camera_to_lidar`, it will change the order of dY and dZ even though your calibration matrix is identical.
> 
> What I am concerned about most is whether they use Lidar coordinate or Camera coordinate in the network during training/inference, especially for functions like `assigned_targets`. So far, I've checked that they use **Lidar** coordinate in CenterPoint, but I can't find codes that they transform the boxes to **Camera** coordinate.

What changes have you made so that your AP is not 0

DJNing, 2022-03-29:
> the same problem, there are some problems when evaluating my data set. My AP and other accuracy are 0. At the same time, I set the 2D information to car, 0,0,0,1,2,3,4 , x. Y, Z, Dy, DX, DZ, rotation angle,according to the issue

Actually, if your dt_annos and gt_annos are exactly the same, it should give you the right answer without any changes

DJNing, 2022-03-29:
> the same problem, there are some problems when evaluating my data set. My AP and other accuracy are 0. At the same time, I set the 2D information to car, 0,0,0,1,2,3,4 , x. Y, Z, Dy, DX, DZ, rotation angle,according to the issue

If you really want to calculate the iou in the lidar coordinate, you can modify the ```d3_box_overlap```, the rinc means the BEV iou. So, change the input parameter of the right axis for function ```rotate_iou_gpu_eval``` to box[:, [0,1,3,4,6]]

laiyingxin2, 2022-03-29:
> 
I made the change according to your method and printed out riou, and the result showed that the value was 0, but when I used demo.py, when visualizing, it is found that there is partial coincidence, and the result AP is 0, which is unreasonable

laiyingxin2, 2022-03-29:
python3 test.py --cfg_file cfgs/kitti_models/voxel_rcnn_car.yaml   --batch_size 3  --ckpt voxel_rcnn_car_84.pth
2022-03-29 19:09:15,841   INFO  **********************Start logging**********************
2022-03-29 19:09:15,842   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-03-29 19:09:15,842   INFO  cfg_file         cfgs/kitti_models/voxel_rcnn_car.yaml
2022-03-29 19:09:15,842   INFO  batch_size       3
2022-03-29 19:09:15,842   INFO  workers          4
2022-03-29 19:09:15,842   INFO  extra_tag        default
2022-03-29 19:09:15,842   INFO  ckpt             voxel_rcnn_car_84.pth
2022-03-29 19:09:15,842   INFO  launcher         none
2022-03-29 19:09:15,842   INFO  tcp_port         18888
2022-03-29 19:09:15,842   INFO  local_rank       0
2022-03-29 19:09:15,842   INFO  set_cfgs         None
2022-03-29 19:09:15,842   INFO  max_waiting_mins 30
2022-03-29 19:09:15,842   INFO  start_epoch      0
2022-03-29 19:09:15,842   INFO  eval_tag         default
2022-03-29 19:09:15,842   INFO  eval_all         False
2022-03-29 19:09:15,842   INFO  ckpt_dir         None
2022-03-29 19:09:15,842   INFO  save_to_file     False
2022-03-29 19:09:15,842   INFO  cfg.ROOT_DIR: /opt/tiger/OpenPCDet
2022-03-29 19:09:15,842   INFO  cfg.LOCAL_RANK: 0
2022-03-29 19:09:15,842   INFO  cfg.CLASS_NAMES: ['Car']
2022-03-29 19:09:15,842   INFO  
cfg.DATA_CONFIG = edict()
2022-03-29 19:09:15,842   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2022-03-29 19:09:15,842   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2022-03-29 19:09:15,842   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2022-03-29 19:09:15,842   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-03-29 19:09:15,842   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2022-03-29 19:09:15,842   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-03-29 19:09:15,842   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2022-03-29 19:09:15,842   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2022-03-29 19:09:15,843   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5']}, 'SAMPLE_GROUPS': ['Car:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2022-03-29 19:09:15,843   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2022-03-29 19:09:15,843   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2022-03-29 19:09:15,843   INFO  
cfg.MODEL = edict()
2022-03-29 19:09:15,843   INFO  cfg.MODEL.NAME: VoxelRCNN
2022-03-29 19:09:15,843   INFO  
cfg.MODEL.VFE = edict()
2022-03-29 19:09:15,843   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2022-03-29 19:09:15,843   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2022-03-29 19:09:15,843   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2022-03-29 19:09:15,843   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
2022-03-29 19:09:15,843   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2022-03-29 19:09:15,843   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2022-03-29 19:09:15,843   INFO  
cfg.MODEL.BACKBONE_2D = edict()
2022-03-29 19:09:15,843   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2022-03-29 19:09:15,843   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5]
2022-03-29 19:09:15,843   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2]
2022-03-29 19:09:15,843   INFO  cfg.MODEL.BACKBONE_2D.NUM_FILTERS: [64, 128]
2022-03-29 19:09:15,843   INFO  cfg.MODEL.BACKBONE_2D.UPSAMPLE_STRIDES: [1, 2]
2022-03-29 19:09:15,843   INFO  cfg.MODEL.BACKBONE_2D.NUM_UPSAMPLE_FILTERS: [128, 128]
2022-03-29 19:09:15,843   INFO  
cfg.MODEL.DENSE_HEAD = edict()
[... 168 lines cut ...]

  state.func_ir.loc))
2022-03-29 19:09:40,065   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00

2022-03-29 19:09:40,066   INFO  Result is save to /opt/tiger/OpenPCDet/output/kitti_models/voxel_rcnn_car/default/eval/epoch_84/val/default
2022-03-29 19:09:40,066   INFO  ****************Evaluation done.*****************

DJNing, 2022-03-29:
> > 
> 
> I made the change according to your method and printed out riou, and the result showed that the value was 0, but when I used demo.py, when visualizing, it is found that there is partial coincidence, and the result AP is 0, which is unreasonable

> > the same problem, there are some problems when evaluating my data set. My AP and other accuracy are 0. At the same time, I set the 2D information to car, 0,0,0,1,2,3,4 , x. Y, Z, Dy, DX, DZ, rotation angle,according to the issue
> 
> If you really want to calculate the iou in the lidar coordinate, you can modify the `d3_box_overlap`, the rinc means the BEV overlaps. So, change the input parameter of the right axis for function `rotate_iou_gpu_eval` to box[:, [0,1,3,4,6]]

Also change the ```d3_box_overlap_kernel``` to use the correct coordinate for height overlap calculation

laiyingxin2, 2022-03-29:
> Also change the `d3_box_overlap_kernel` to use the correct coordinate for height overlap calculation

Do you mean coordinate conversion? I tried to change the X, y and Z of the label. It seems to have no effect，What should I do to change

def bev_box_overlap(boxes, qboxes, criterion=-1):
    riou = rotate_iou_gpu_eval(boxes[:, [0,1,2,3,4]], qboxes, criterion)
    print("riou")
    print(riou)
    return riou

@numba.jit(nopython=True, parallel=True)
def d3_box_overlap_kernel(boxes, qboxes, rinc, criterion=-1):
    # ONLY support overlap in CAMERA, not lider.
    #boxes=boxes[:,[0,1,2,3,4]]
    print("rinc")
    print(rinc)
    N, K = boxes.shape[0], qboxes.shape[0]
    for i in range(N):
        for j in range(K):
            if rinc[i, j] > 0:
#                 iw = (min(boxes[i, 1] + boxes[i, 4], qboxes[j, 1] +
#                         qboxes[j, 4]) - max(boxes[i, 1], qboxes[j, 1]))
                iw = (min(boxes[i, 1], qboxes[j, 1]) - max(
                    boxes[i, 1] - boxes[i, 4], qboxes[j, 1] - qboxes[j, 4]))

                if iw >= 0:
                    area1 = boxes[i, 3] * boxes[i, 4] * boxes[i, 5]
                    area2 = qboxes[j, 3] * qboxes[j, 4] * qboxes[j, 5]
                    inc = iw * rinc[i, j]
                    if criterion == -1:
                        ua = (area1 + area2 - inc)
                    elif criterion == 0:
                        ua = area1
                    elif criterion == 1:
                        ua = area2
                    else:
                        ua = inc
                    rinc[i, j] = inc / ua
                else:
                    rinc[i, j] = 0.0

def d3_box_overlap(boxes, qboxes, criterion=-1):
    rinc = rotate_iou_gpu_eval(boxes[:, [0, 2, 3, 5, 6]],
                               qboxes[:, [0, 2, 3, 5, 6]], 2)
    d3_box_overlap_kernel(boxes, qboxes, rinc, criterion)
    return rinc

laiyingxin2, 2022-03-29:
`def bev_box_overlap(boxes, qboxes, criterion=-1):
    riou = rotate_iou_gpu_eval(boxes[:, [0,1,2,3,4]], qboxes, criterion)
    print("riou")
    print(riou)
    return riou

@numba.jit(nopython=True, parallel=True)
def d3_box_overlap_kernel(boxes, qboxes, rinc, criterion=-1):
    # ONLY support overlap in CAMERA, not lider.
    #boxes=boxes[:,[0,1,2,3,4]]
    print("rinc")
    print(rinc)
    N, K = boxes.shape[0], qboxes.shape[0]
    for i in range(N):
        for j in range(K):
            if rinc[i, j] > 0:
#                 iw = (min(boxes[i, 1] + boxes[i, 4], qboxes[j, 1] +
#                         qboxes[j, 4]) - max(boxes[i, 1], qboxes[j, 1]))
                iw = (min(boxes[i, 1], qboxes[j, 1]) - max(
                    boxes[i, 1] - boxes[i, 4], qboxes[j, 1] - qboxes[j, 4]))

                if iw >= 0:
                    area1 = boxes[i, 3] * boxes[i, 4] * boxes[i, 5]
                    area2 = qboxes[j, 3] * qboxes[j, 4] * qboxes[j, 5]
                    inc = iw * rinc[i, j]
                    if criterion == -1:
                        ua = (area1 + area2 - inc)
                    elif criterion == 0:
                        ua = area1
                    elif criterion == 1:
                        ua = area2
                    else:
                        ua = inc
                    rinc[i, j] = inc / ua
                else:
                    rinc[i, j] = 0.0

def d3_box_overlap(boxes, qboxes, criterion=-1):
    rinc = rotate_iou_gpu_eval(boxes[:, [0, 2, 3, 5, 6]],
                               qboxes[:, [0, 2, 3, 5, 6]], 2)
    d3_box_overlap_kernel(boxes, qboxes, rinc, criterion)
    return rinc`

laiyingxin2, 2022-03-29:
> Also change the `d3_box_overlap_kernel` to use the correct coordinate for height overlap calculation

Thank you for your reply. If successful, I will synchronize on this issue.

DJNing, 2022-03-29:
> Thank you for your reply. If successful, I will synchronize on this issue.

Note that the kitti dataset will convert the lidar bbox to camera coordinate, so the evaluation code is correct for kitti dataset

laiyingxin2, 2022-03-30:
Now I feel very confused. After coordinate conversion, I have visualized my data set and label, which can be matched, but the accuracy is still 0. At the same time, the code converted to the camera coordinate system should be in pcdet / dataset / Kitti / Kitti_utils.py file

closed 2022-05-15 by github-actions[bot]