open-mmlab/OpenPCDet #913: inference output format

YCAyca, 2022-03-29:
Hello, when we run demo.py, we call pred_dicts, _ = model.forward(data_dict) and print(pred_dicts[0]['pred_boxes']) gives us the predicted boxes for 1 scene. The sample output for this is like below:

[image]

So we have 8 coordinates for each box. What is the order of these coordinates? x,y,z,rot_y,w,h,l or x,y,z,w,h,l,rot_y maybe?

fjczx, 2022-03-29:
I found this in open3d_vis_utils.py
```
def translate_boxes_to_open3d_instance(gt_boxes):
    center = gt_boxes[0:3]
    lwh = gt_boxes[3:6]
    axis_angles = np.array([0, 0, gt_boxes[6] + 1e-10])
    rot = open3d.geometry.get_rotation_matrix_from_axis_angle(axis_angles)
```

laiyingxin2, 2022-03-29:
They are 3D center coordinates and length, width and height respectively

laiyingxin2, 2022-03-29:
> They are 3D center coordinates and length, width and height respectively

By the way, the last one should be the rotation angle

YCAyca, 2022-03-29:
Yes I checked with the results and its true. both ground truth boxes and prediction boxes comes with the same format as x,y,z,l,w,h,rot_y Thank you!!

closed 2022-03-29 by YCAyca