open-mmlab/OpenPCDet #916: Concatenate datasets

BrokenVoodooDoll, 2022-04-04:
Is it possible to learn models in OpenPCDet using several datasets simultaneously (like using ConcatDataset in pytorch or in mmdetection3d)?

laiyingxin2, 2022-04-07:
I think converting other data into Kitti format can be successful, but no one should have tried it on openpcdet

sshaoshuai, 2022-04-14:
We still do not have this feature in OpenPCDet, theoretically all the datasets have already been converted to our unified coordinate system in their own data processing function, and what we need to do is to add a ConcatDataset.
Since currently different datasets have different data patterns, so I think it is not that important to add this feature, but PR is also welcomed to add it.

closed 2022-05-29 by github-actions[bot]