open-mmlab/OpenPCDet #923: from . import iou3d_nms_cuda

alanWXZ, 2022-04-12:
Traceback (most recent call last):
  File "/media/ExtHDD01/wangxuzhi/code/OpenPCDet-master/tools/train.py", line 9, in <module>
    from test import repeat_eval_ckpt
  File "/media/ExtHDD01/wangxuzhi/code/OpenPCDet-master/tools/test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/media/ExtHDD01/wangxuzhi/code/OpenPCDet-master/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "../pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "../pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    from . import iou3d_nms_cuda
ImportError: ../pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-37m-x86_64-linux-gnu.so: failed to map segment from shared object

alanWXZ, 2022-04-12:
I have successfully installed the openpcdet library. However, when I run train.py, the above error arises.

closed 2022-05-28 by github-actions[bot]

RakshiShreya, 2022-09-13:
> I have successfully installed the openpcdet library. However, when I run train.py, the above error arises.

Did your error has been solved?
I am facing the same issue.  If solved, please do help me

christysch, 2022-10-19:
> 

Hi i found that is because your Pytorch-GPU doesn't match your CUDA version . try to reinstall the properly pytorch-GPU that you will find the error missing.

BraunBenni, 2022-10-23:
> Help~~~ i already install the OpenPCDet ,and i can't find the iou3d_nms_cuda file. Does anyone know how to fix it?

I got the same issue. Does anyone know where this file comes from and what it is for?