open-mmlab/OpenPCDet #928: The feature become Nan after the “conv_input” layer in the "VoxelResBackBone8x"

twn29004, 2022-04-14:
When I was training centernet with my own loss function, I found that in VoxelResBack8x, the features became nan after going through the conv_input layer.

https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/pcdet/models/backbones_3d/spconv_backbone.py#L252-L260

And the conv_input define is:
https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/pcdet/models/backbones_3d/spconv_backbone.py#L191-L195

And I have checked the input voxel_feature is not Nan.  Can anyone give me a suggestion？I don't know what to do with it.  Below are some results from my debugging process.
[image]
[image]

twn29004, 2022-04-14:
[image]

twn29004, 2022-04-14:
My pcdet version is 0.5.1+eed89a4, and the spconv version is spconv-cu114  2.1.21

twn29004, 2022-04-15:
I have found the problem. is the problem with my loss function.

closed 2022-04-15 by twn29004

JianyyuWang, 2022-06-19:
Do you solve this problem？

twn29004, 2022-06-19:
> 
There are some error in my loss function. May be you can add the following code in the training function to check the probelm.
```python
import torch  
# forward 
torch.autograd.set_detect_anomaly(True)  
# TODO
# backward  
with torch.autograd.detect_anomaly():  
    loss.backward()  
```

study1994, 2023-07-17:
[image]
Same problem, following the workaround should not be a loss problem