open-mmlab/OpenPCDet #934: ptxas application ptx input, line 9; fatal   : Unsupported .version 7.3; current version is '7.2'

saurtone, 2022-04-17:
hello,I was testing test.py and then there are some bug ,I can't deal with them ,I want to be helped.

The same problem as this friend had
[https://github.com/open-mmlab/OpenPCDet/issues/675](url)

luminous33, 2022-05-04:
It can be solved by matching the CUDA version and the NVIDIA driver version together.
I also met the same error.
I'm using cuda-11.3. 
before nvidia-driver = NVIDIA-Linux-x86_64-460.91.03
but now my nvidia-driver = NVIDIA-Linux-x86_64-465.19.01
So i solved that error.
rtx-3090, python=3.9, cuda-11.3(my env info)

saurtone, 2022-05-04:
> It can be solved by matching the CUDA version and the NVIDIA driver version together. I also met the same error. I'm using cuda-11.3. before nvidia-driver = NVIDIA-Linux-x86_64-460.91.03 but now my [...]

Thank you for your answer! I tried all CUDA versions and numba versions, but I didn't find the answer. Maybe the problem lies in the drive. I'll tell you the result after I try to change the driver. Please wait

saurtone, 2022-05-05:
> Thank you for your answer! I tried all CUDA versions and numba versions, but I didn't find the answer. Maybe the problem lies in the drive. I'll tell you the result after I try to change the driver. [...]

> It can be solved by matching the CUDA version and the NVIDIA driver version together. I also met the same error. I'm using cuda-11.3. before nvidia-driver = NVIDIA-Linux-x86_64-460.91.03 but now my [...]
 i solved that error rtx-2080ti,python=3.8 cuda-10.2 nvidia-driver = NVIDIA-Linux-x86_64-460.67

Eaphan, 2022-05-12:
I solved the problem by installing cudatoolkit.
conda install -c numba cudatoolkit=x.x  (8.0, 9.0, 10.0, depend on your environment)

closed 2022-06-27 by github-actions[bot]

yuwenmichael, 2023-09-11:
> I solved the problem by installing cudatoolkit. conda install -c numba cudatoolkit=x.x (8.0, 9.0, 10.0, depend on your environment)

For me is to run with `conda install -c numba cudatoolkit=11.0`.
Followed by this link: https://github.com/GoogleCloudPlatform/nvidia-merlin-on-vertex-ai/issues/27,  I found that the `CUDA Device Initialized Version` and `CUDA Runtime Version` should be the same by running `numba -s | grep CUDA`. Since I am having CUDA==11.0 on `nvidia-smi`, installing cudatoolkit=11.0 works for me.
Hope this help)

saurtone, 2023-09-11:
邮件我已收到。辛苦了。