open-mmlab/OpenPCDet #935: How to convert openpcd coordinate system to Kitti coordinate system

CUIMD, 2022-04-17:
Not only the transformation of format, but also the mapping of coordinate system. I want to train my own point cloud data with Kitti parameters. My own point cloud data is in the format of openpcd [x, y, z, Dx, Dy, Dz, R] and the mapping of Kitti coordinate system is not good

shaoxh, 2022-05-19:
A function in demo.py like that as follows.
def translate_boxes_to_open3d_instance(gt_boxes):
    """
             4-------- 6
           /|         /|
          5 -------- 3 .
          | |        | |
          . 7 -------- 1
          |/         |/
          2 -------- 0
    """
I think a open3d box could be transed to a kitti output.
Looking for the solution on my own.

shaoxh, 2022-05-19:
Some more infomation.
`def translate_boxes_to_open3d_instance(gt_boxes):
    """
             4-------- 6
           /|         /|
          5 -------- 3 .
          | |        | |
          . 7 -------- 1
          |/         |/
          2 -------- 0
    """
    center = gt_boxes[0:3]
    lwh = gt_boxes[3:6]
    axis_angles = np.array([0, 0, gt_boxes[6] + 1e-10])
    rot = open3d.geometry.get_rotation_matrix_from_axis_angle(axis_angles)
    box3d = open3d.geometry.OrientedBoundingBox(center, rot, lwh)

    line_set = open3d.geometry.LineSet.create_from_oriented_bounding_box(box3d)

    # import ipdb; ipdb.set_trace(context=20)
    lines = np.asarray(line_set.lines)
    lines = np.concatenate([lines, np.array([[1, 4], [7, 6]])], axis=0)

    line_set.lines = open3d.utility.Vector2iVector(lines)

    return line_set, box3d`

closed 2022-07-04 by github-actions[bot]

barkabi, 2024-07-30:
> Some more infomation. `def translate_boxes_to_open3d_instance(gt_boxes): center = gt_boxes[0:3] lwh = gt_boxes[3:6] axis_angles = np.array([0, 0, gt_boxes[6] + 1e-10]) rot = open3d.geometry.get_rotation_matrix_from_axis_angle(axis_angles) box3d = open3d.geometry.OrientedBoundingBox(center, rot, lwh)
> 
> ```
> line_set = open3d.geometry.LineSet.create_from_oriented_bounding_box(box3d)
> 
> # import ipdb; ipdb.set_trace(context=20)
> lines = np.asarray(line_set.lines)
> lines = np.concatenate([lines, np.array([[1, 4], [7, 6]])], axis=0)
> 
> line_set.lines = open3d.utility.Vector2iVector(lines)
> 
> return line_set, box3d`
> ```

It seems that It's not kitti format. The values: (line_set, box3d) are just:  LineSet with 14 lines. OrientedBoundingBox: center: (60.138, -11.7918, 0.160769), extent: 4.14645, 1.67387, 1.52242)