open-mmlab/OpenPCDet #938: Multi GPU training in the first epoch blocking problem

exiawsh, 2022-04-19:
Hello, when I use multi GPU training, if the number of workers is set to more than 1, the training will be stuck at the beginning of the first epoch. But if it is a single GPU, i can set the number of workers more than one. Have you ever encountered such a problem? How can I solve it?

sshaoshuai, 2022-04-19:
Have you tried to comment these two lines to quickly start the training? 

https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/pcdet/utils/common_utils.py#L162-L163

exiawsh, 2022-04-20:
> Have you tried to comment these two lines to quickly start the training?
> 
> https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/pcdet/utils/common_utils.py#L162-L163
Thank you for the reply. I will try it now. But I have I have another question about the waymo dataset. Because the waymo dataset is huge. 
After I process the data and I want to migrate datasets to another disk, could I just migrate the  generated data?
That's mean, could I manage the data like this:
OpenPCDet
├── data
│   ├── waymo
│   │   │── ImageSets
|   |   |── waymo_processed_data_v0_5_0
│   │   │   │── segment-xxxxxxxx/
|   |   |   |── ...
│   │   │── waymo_processed_data_v0_5_0_gt_database_train_sampled_1/
│   │   │── waymo_processed_data_v0_5_0_waymo_dbinfos_train_sampled_1.pkl
│   │   │── waymo_processed_data_v0_5_0_gt_database_train_sampled_1_global.npy (optional)
│   │   │── waymo_processed_data_v0_5_0_infos_train.pkl (optional)
│   │   │── waymo_processed_data_v0_5_0_infos_val.pkl (optional)
├── pcdet
├── tools
just delete the raw_data and use the processed_data

sshaoshuai, 2022-04-20:
Sure, the processed data is already enough for training and evaluation.

exiawsh, 2022-04-20:
> Have you tried to comment these two lines to quickly start the training?
> 
> https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/pcdet/utils/common_utils.py#L162-L163

Thank you, it works!
And if I processed the waymo dataset correctly, how much disk memory would be occupied by the processed data?

sshaoshuai, 2022-04-20:
Maybe 300-400GB, I do not remember the exact number.

closed 2022-04-23 by exiawsh

jiaminglei-lei, 2022-05-26:
> Have you tried to comment these two lines to quickly start the training?
> 
> https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/pcdet/utils/common_utils.py#L162-L163

I have the same problem: multi-gpu training will be stuck at the beginning, and **the machine memory will increase gradually** until the error : `RuntimeError: DataLoader worker (pid 174116) is killed by signal: Killed.` happen.
And uncomment these lines, it fixed. I am curious about why it works?