open-mmlab/OpenPCDet #942: the error about generating the infos of waymo dataset

Feng-HC, 2022-04-20:
When I try to generate the information of waymo dataset following your instructions in `getting started`,  I faced the issue as shown below.
```python
2022-04-20 12:20:34,329   INFO  Loading Waymo dataset
2022-04-20 12:20:34,330   INFO  Total skipped info 5
2022-04-20 12:20:34,330   INFO  Total samples for Waymo dataset: 0
2022-04-20 12:20:34,330   INFO  Loading Waymo dataset
2022-04-20 12:20:34,331   INFO  Total skipped info 20
2022-04-20 12:20:34,331   INFO  Total samples for Waymo dataset: 0
2022-04-20 12:20:35.347072: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcuda.so.1
2022-04-20 12:20:35.349653: E tensorflow/stream_executor/cuda/cuda_driver.cc:313] failed call to cuInit: CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected
2022-04-20 12:20:35.349680: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:169] retrieving CUDA diagnostic information for host: fhc-HP
2022-04-20 12:20:35.349685: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:176] hostname: fhc-HP
2022-04-20 12:20:35.349760: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:200] libcuda reported version is: 440.44.0
2022-04-20 12:20:35.349780: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:204] kernel reported version is: 440.44.0
2022-04-20 12:20:35.349784: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:310] kernel version seems to match DSO: 440.44.0
2022-04-20 12:20:35.349966: I tensorflow/core/platform/cpu_feature_guard.cc:143] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 AVX512F FMA
---------------The waymo sample interval is 1, total sequecnes is 20-----------------
2022-04-20 12:20:35.378346: I tensorflow/core/platform/profile_utils/cpu_utils.cc:102] CPU Frequency: 3800000000 Hz
2022-04-20 12:20:35.379059: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x55fdc7856530 initialized for platform Host (this does not guarantee that XLA will be used). Devices:
2022-04-20 12:20:35.379099: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Host, Default Version
```

My device info as show below:
GPU : RTX2080Ti
cuda : 10.2
cudnn : 7.6.5
nvidia driver : 440.44
Tensorflow : 2.2.0
waymo-open-dataset-tf-2-2-0

Is it because my versions of tensorflow and cuda too low to generate the info of waymo dataset?

If so, Could you tell me your versions of cuda, cudnn, nvidia driver and tensorflow?

sshaoshuai, 2022-04-22:
You can check the directory `OpenPCDet/data/waymo` to see whether there are new files, since I guess it is in the process of generating infos.

Feng-HC, 2022-04-23:
I have checked the directory `OpenPCDet/data/waymo`, and find it is only generating **part of data infos** of `train set` in the directory `waymo_processed_data_v0_5_0`. And the rest of data infos such as `waymo_processed_****.infors_train.pkl` are still not generated.  The program seem to stop in line 152 of `waymo_dataset.py` as shown below.
```python
def get_infos(self, ...)
    ***

    with multiprocessing.Pool(num_workers) as p:
        sequence_infos = list(tqdm(p.imap(process_single_sequence, sample_sequence_file_list), total=len(sample_sequence_file_list)))
```

Feng-HC, 2022-04-23:
I have checked the directory `OpenPCDet/data/waymo`, and find it is only generating **part of data infos** of `train set` in the directory `waymo_processed_data_v0_5_0`. And the rest of data infos such as `waymo_processed_****.infors_train.pkl` are still not generated.  The program seem to stop in line 152 of `waymo_dataset.py` as shown below.
```python
def get_infos(self, ...)
    ***

    with multiprocessing.Pool(num_workers) as p:
        sequence_infos = list(tqdm(p.imap(process_single_sequence, sample_sequence_file_list), total=len(sample_sequence_file_list)))
```

> You can check the directory `OpenPCDet/data/waymo` to see whether there are new files, since I guess it is in the process of generating infos.

sshaoshuai, 2022-04-24:
Sorry, no idea about what happends, but generally it will take several minutes to see the pkl file in each segment folder (There will first be some npy files and each segment folder will have its own pkl file).  

Maybe you can try to call function `process_single_sequence` separately instead of using multiprocessing, to see whether it works.

Feng-HC, 2022-04-24:
Is it because my versions of tensorflow and cuda too low to generate the info of waymo dataset?
Could you tell me your versions of cuda and tensorflow?

sshaoshuai, 2022-05-03:
I use the following command to install waymo dataset and tensorflow:
```pip3 install waymo-open-dataset-tf-2-5-0```

closed 2022-06-17 by github-actions[bot]