open-mmlab/OpenPCDet #945: train  custom dataset  ,occur thins error

ccsself, 2022-04-22:
N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0

OrangeSodahub, 2022-05-03:
Hello, I occurred this now, too. Have you solved the problem?

rikardlegge-einride, 2022-05-04:
Also encountered this issue, one of the culprits I found was `transform_points_to_voxels` in `data_processor.py` producing voxels with 0 points in them, later triggering the assertion.

My hypothesis, which I have not confirmed yet, is that augmentations and sampling sometimes thins out the point cloud enough to cause this invalid state. The hypothesis is based on bumping into this issue when trying to train some of the models using lower resolution point clouds than can be found in the KITTI dataset. 

In my case this was triggered around once every 10000 samples and I therefore added two points of semi-random data (which I would not recommend). It at least fixed the assertion without having to dig into the internals of the voxel generator or the rest of the system.

If someone else has a better solution, I would love to hear about it!

ccsself, 2022-05-17:
> Hello, I occurred this now, too. Have you solved the problem?

it hasn't been solved yet,  i   don't konw  how to adjust the parameter

closed 2022-06-30 by github-actions[bot]