open-mmlab/OpenPCDet #946: Visulization seems not right

mc171819, 2022-04-22:
Hi, I use demo.py to visulise the result predicted by centerpoint, it seems not right:
[image]
The accuracy I get is:
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7142 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7090 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6327 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6280 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.7284 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.6672 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.6437 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.5884 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000 
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.7053 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.6931 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.6795 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.6677 

How can this happen?

mc171819, 2022-04-22:
By the way, the sequece i use for vis is segment-1005081002024129653_5313_150_5333_150_with_camera_labels

sshaoshuai, 2022-04-22:
I guess it's the problem of visualization: 
https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/tools/visual_utils/visualize_utils.py#L163-L168

By default we only show 100 boxes at most by considering efficiency, and you can increase this number by yourself.

mc171819, 2022-04-23:
Solved

closed 2022-04-23 by mc171819

bigsheep2012, 2022-04-25:
Hello @mc171819,

May I know how you sovle it? and are u suing the default configuration(centerpoint.yaml) to train center point for 30 epoch(interval=5) on waymo dataset?

bigsheep2012, 2022-04-27:
hi @mc171819,
use the author's implementation of WaymoDataset solve the issue. Thanks.

717770780, 2023-08-08:
> 
May I know how you sovle it? Did you just replace DemoDataset with WaymoDataset? Thanks.

717770780, 2023-08-08:
> hi @mc171819, use the author's implementation of WaymoDataset solve the issue. Thanks.

May I know how you sovle it? Did you just replace DemoDataset with WaymoDataset? Thanks.

CaiFuyao, 2024-05-13:
> > hi @mc171819, use the author's implementation of WaymoDataset solve the issue. Thanks.
> 
> May I know how you sovle it? Did you just replace DemoDataset with WaymoDataset? Thanks.
Have you ever sovled this problem？Thanks!

CaiFuyao, 2024-05-13:
May I know how you solve it? I change the __getitem__() method but it dosen't work. Could you show me your code? Thanks!!!