open-mmlab/OpenPCDet #949: Wrong Centerpoint demo result and eval error

bigsheep2012, 2022-04-24:
Hi openpcdet team @cgerum @lindahua @chreisinger @acivgin1 @sshaoshuai , 
Thanks for sharing this framework.

I follow the getting_start.md, my environment setups(part) is as follows:

pytorch 1.11.0
cuda 11.3
spconv 2.1.21
numba: 0.55.1
open3d 0.15.2
waymo: 1_3_0
waymo-open-dataset-tf-2-5-0 1.4.1
gpu: 3090 *1

I pulled the newest code of pcdet repo and tried to reproduce the mentioned centerpoint result here:
[CenterPoint](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/waymo_models/centerpoint_without_resnet.yaml)	71.33/70.76	63.16/62.65	72.09/65.49	64.27/58.23	68.68/67.39	66.11/64.87

I did the following tests:
1) I did not modify any code, and trained centerpoint for 30 epoch(with default centerpoint.yaml)
python train.py --cfg_file cfg_file cfgs/waymo_models/centerpoint.yaml

2) tried the demo.py with the following command 
**python demo.py --cfg_file cfgs/waymo_models/centerpoint.yaml --ckpt ../output/waymo_models/centerpoint/default/ckpt/checkpoint_epoch_30.pth --data_path ../data/waymo/waymo_processed_data_v0_5_0/segmentxxx/0000.npy --ext '.npy'**

**The demo result seems incorrect**:
[image]

3) tried the test.py
**python test.py --cfg_file cfgs/waymo_models/centerpoint.yaml --ckpt ../output/waymo_models/centerpoint/default/ckpt/checkpoint_epoch_30.pth --batch_size 4**

get logs(full log is attached) and erros (full log
[log.txt](https://github.com/open-mmlab/OpenPCDet/files/8549254/log.txt)
 is attached):
2022-04-24 15:15:00,441   INFO  *************** EPOCH 30 EVALUATION *****************
eval:   0%|                                                                                                                                    | 0/9997 [00:00<?, ?it/s]/data/lin/research/github/OpenPCDet/tools/../pcdet/models/model_utils/centernet_utils.py:142: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  topk_ys = (topk_inds // width).float()
/data/lin/research/github/OpenPCDet/tools/../pcdet/models/model_utils/centernet_utils.py:146: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  topk_classes = (topk_ind // K).int()
eval: 100%|██████████████████████████████████████████████████████████████████████████████████████| 9997/9997 [27:20<00:00,  6.09it/s, recall_0.3=(0, 1541320) / 1800720]
2022-04-24 15:42:21,291   INFO  *************** Performance of EPOCH 30 *****************
2022-04-24 15:42:21,301   INFO  Generate label finished(sec_per_example: 0.0410 second).
2022-04-24 15:42:21,302   INFO  recall_roi_0.3: 0.000000
2022-04-24 15:42:21,302   INFO  recall_rcnn_0.3: 0.855947
2022-04-24 15:42:21,302   INFO  recall_roi_0.5: 0.000000
2022-04-24 15:42:21,302   INFO  recall_rcnn_0.5: 0.768198
2022-04-24 15:42:21,302   INFO  recall_roi_0.7: 0.000000
2022-04-24 15:42:21,302   INFO  recall_rcnn_0.7: 0.503220
2022-04-24 15:42:21,353   INFO  Average predicted number of objects(39987 samples): 111.948
2022-04-24 15:42:24.638565: I tensorflow/stream_executor/platform/default/dso_loader.cc:53] Successfully opened dynamic library libcudart.so.11.0
Start the waymo evaluation...
Number: (pd, 4476454) VS. (gt, 1630510)
Level 1: 1377451, Level2: 253059)
Traceback (most recent call last):
  File "test.py", line 200, in <module>
    main()
  File "test.py", line 196, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 62, in eval_single_ckpt
    eval_utils.eval_one_epoch(
  File "/data/lin/research/github/OpenPCDet/tools/eval_utils/eval_utils.py", line 110, in eval_one_epoch
    result_str, result_dict = dataset.evaluation(
  File "/data/lin/research/github/OpenPCDet/tools/../pcdet/datasets/waymo/waymo_dataset.py", line 315, in evaluation
    ap_result_str, ap_dict = waymo_eval(eval_det_annos, eval_gt_annos)
  File "/data/lin/research/github/OpenPCDet/tools/../pcdet/datasets/waymo/waymo_dataset.py", line 298, in waymo_eval
    ap_dict = eval.waymo_evaluation(
  File "/data/lin/research/github/OpenPCDet/tools/../pcdet/datasets/waymo/waymo_eval.py", line 206, in waymo_evaluation
    metrics = self.build_graph(graph)
  File "/data/lin/research/github/OpenPCDet/tools/../pcdet/datasets/waymo/waymo_eval.py", line 122, in build_graph
    metrics = detection_metrics.get_detection_metric_ops(
  File "/home/server/.local/lib/python3.8/site-packages/waymo_open_dataset/metrics/python/detection_metrics.py", line 134, in get_detection_metric_ops
    ground_truth_speed = tf.zeros((num_gt_boxes, 2), tf.float32)
  File "/home/server/.local/lib/python3.8/site-packages/tensorflow/python/util/dispatch.py", line 206, in wrapper
    return target(*args, **kwargs)
  File "/home/server/.local/lib/python3.8/site-packages/tensorflow/python/ops/array_ops.py", line 2911, in wrapped
    tensor = fun(*args, **kwargs)
  File "/home/server/.local/lib/python3.8/site-packages/tensorflow/python/ops/array_ops.py", line 2960, in zeros
    output = _constant_if_small(zero, shape, dtype, name)
  File "/home/server/.local/lib/python3.8/site-packages/tensorflow/python/ops/array_ops.py", line 2896, in _constant_if_small
    if np.prod(shape) < 1000:
  File "<__array_function__ internals>", line 5, in prod
  File "/home/server/anaconda3/envs/torch11/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 3051, in prod
    return _wrapreduction(a, np.multiply, 'prod', axis, dtype, out,
  File "/home/server/anaconda3/envs/torch11/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 86, in _wrapreduction
    return ufunc.reduce(obj, axis, dtype, out, **passkwargs)
  File "/home/server/.local/lib/python3.8/site-packages/tensorflow/python/framework/ops.py", line 867, in __array__
    raise NotImplementedError(
NotImplementedError: Cannot convert a symbolic Tensor (strided_slice:0) to a numpy array. This error may indicate that you're trying to pass a Tensor to a NumPy call, which is not supported

The tensorbaord visualization seems correct:
[image]

I tried the pretrained second model on kitti dataset, the demo result seems correct.
Do I miss anything (that lead to the wrong demo result)? 

Thanks in advance,
Lin

MartinHahner, 2022-04-26:
[demo.py](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/demo.py) uses a "lightweight" `DemoDataset`
https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/tools/demo.py#L23

Replace the `DemoDataset` with the "original" `WaymoDataset` and it should work.
https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/pcdet/datasets/waymo/waymo_dataset.py#L21

bigsheep2012, 2022-04-27:
hello @MartinHahner ,
Use the WaymoDataset sovle the demo problem.   The **error**("**NotImplementedError: Cannot convert a symbolic Tensor (strided_slice:0) to a numpy array**")  is solved by downgrading numpy from 1.21.1 to 1.19.2.  I guess it's caused by the package waymo-open-dataset-tf-2-5-0 (possibly not compatible with python3.8 and numpy 1.21). There is a newer one: waymo-open-dataset-tf-2-6-0. maybe update to this version solving it automatically.

closed 2022-04-27 by bigsheep2012

SkywalkerHe, 2022-11-29:
Hello @bigsheep2012 ,
This is my configuration：
numba    0.55.1
numpy    1.21.0
torch       1.11.0+cu113
waymo-open-dataset-tf-2-6-0    1.4.9
GPU    A100*8
When I run 
`bash scripts/dist_test.sh ${NUM_GPUS}  --cfg_file cfgs/waymo_models/centerpoint_4frames.yaml \
--ckpt ../output/waymo_models/centerpoint_4frames/default/ckpt/checkpoint_epoch_36.pth \
--set DATA_CONFIG.DATA_SPLIT.test train
`
use torch1.13+numpy1.23.5, I got same error with, but I use this configuration I got other error:
RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xe
Traceback (most recent call last):
  File "test.py", line 14, in <module>
RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xe
RuntimeErrorTraceback (most recent call last):
: module compiled against API version 0x10 but this version of numpy is 0xe  File "test.py", line 14, in <module>

Traceback (most recent call last):
  File "test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/home/jovyan/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
    from eval_utils import eval_utils
  File "/home/jovyan/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
    from eval_utils import eval_utils
  File "/home/jovyan/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "/home/jovyan/OpenPCDet/tools/../pcdet/models/__init__.py", line 6, in <module>
    from pcdet.models import load_data_to_gpu
  File "/home/jovyan/OpenPCDet/tools/../pcdet/models/__init__.py", line 6, in <module>
    from pcdet.models import load_data_to_gpu
  File "/home/jovyan/OpenPCDet/tools/../pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/home/jovyan/OpenPCDet/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detectors import build_detector
  File "/home/jovyan/OpenPCDet/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detectors import build_detector
  File "/home/jovyan/OpenPCDet/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/home/jovyan/OpenPCDet/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/home/jovyan/OpenPCDet/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/home/jovyan/OpenPCDet/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/jovyan/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 8, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/jovyan/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 8, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/jovyan/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 8, in <module>
    from ...utils import common_utils
  File "/home/jovyan/OpenPCDet/tools/../pcdet/utils/common_utils.py", line 7, in <module>
    from ...utils import common_utils
  File "/home/jovyan/OpenPCDet/tools/../pcdet/utils/common_utils.py", line 7, in <module>
    from ...utils import common_utils
  File "/home/jovyan/OpenPCDet/tools/../pcdet/utils/common_utils.py", line 7, in <module>
    import SharedArray
ImportError: numpy.core.multiarray failed to import
RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xe
Traceback (most recent call last):
  File "test.py", line 14, in <module>
    import SharedArray
ImportError: numpy.core.multiarray failed to import
    import SharedArray
ImportError: numpy.core.multiarray failed to import
    from eval_utils import eval_utils
  File "/home/jovyan/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
[... 73 lines cut ...]
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/jovyan/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 8, in <module>
    from ...utils import common_utils
  File "/home/jovyan/OpenPCDet/tools/../pcdet/utils/common_utils.py", line 7, in <module>
    import SharedArray
ImportError: numpy.core.multiarray failed to import
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 42457) of binary: /opt/conda/bin/python
Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/opt/conda/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/jovyan/.local/lib/python3.8/site-packages/torch/distributed/launch.py", line 193, in <module>
    main()
  File "/home/jovyan/.local/lib/python3.8/site-packages/torch/distributed/launch.py", line 189, in main
    launch(args)
  File "/home/jovyan/.local/lib/python3.8/site-packages/torch/distributed/launch.py", line 174, in launch
    run(args)
  File "/home/jovyan/.local/lib/python3.8/site-packages/torch/distributed/run.py", line 715, in run
    elastic_launch(
  File "/home/jovyan/.local/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 131, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/home/jovyan/.local/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 245, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError:

bigsheep2012, 2022-12-02:
hi @SkywalkerHe 
The sharedarray problem seems like a duplicate issue
Try this solution [here](https://github.com/open-mmlab/OpenPCDet/issues/867#issuecomment-1075166761)

good luck