open-mmlab/OpenPCDet #950: Error in batch_size in PointPillarScatter

GuillaumeAnoufa, 2022-04-26:
Hello, 

I am trying to train a pointpillar model on custom datasets of point clouds. I'm facing an error in forward function of PointPillarScatter class. 
batch_size is defined as `batch_size = coords[:, 0].max().int().item() + 1` instead of `batch_size = batch_dict['batch_size']`. 
Therefore every few iterations`batch_dict['spatial_features']` is of shape `torch.Size([3, 64, 214272])` instead of `torch.Size([4, 64, 214272])`
This results in an error when computing loss: 
`  File "/home/ganoufa/workSpace/deep_learning/lidar_detection/OpenPCDet/tools/../pcdet/utils/loss_utils.py", line 60, in forward
    pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid
RuntimeError: The size of tensor a (4) must match the size of tensor b (3) at non-singleton dimension 0
`
What can be the cause of coords[:, 0].max().int().item() + 1 not being equal to the actual batch_size ?

GuillaumeAnoufa, 2022-04-27:
I am also unable to use other models. I get different error messages that are probably linked with the first issue :
For **pv_rcnn** : in **VoxelBackBone8x** forward function 
I get an error during convolutions
```
tensor shapes:
x_conv1:  [2, 248, 216]
x_conv2:  [1, 124, 108]
x_conv3:  [1, 62, 54]
```
x_conv4 computation results in an error
```

[Exception|implicit_gemm_pair]indices=torch.Size([2557, 4]),bs=2,ss=[1, 62, 54],algo=ConvAlgo.MaskImplicitGemm,ksize=[3, 3, 3],stride=[2, 2, 2],padding=[0, 1, 1],dilation=[1, 1, 1],subm=False,transpose=False
  File "/home/ganoufa/workSpace/deep_learning/lidar_detection/OpenPCDet/tools/../pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/ganoufa/workSpace/deep_learning/AnomalyDetection/myenv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/ganoufa/workSpace/deep_learning/lidar_detection/OpenPCDet/tools/../pcdet/models/backbones_3d/spconv_backbone.py", line 157, in forward
    x_conv4 = self.conv4(x_conv3)

[.........]
  File "/home/ganoufa/workSpace/deep_learning/AnomalyDetection/myenv/lib/python3.8/site-packages/spconv/pytorch/conv.py", line 381, in forward
    res = ops.get_indice_pairs_implicit_gemm(
  File "/home/ganoufa/workSpace/deep_learning/AnomalyDetection/myenv/lib/python3.8/site-packages/spconv/pytorch/ops.py", line 280, in get_indice_pairs_implicit_gemm
    raise ValueError(
ValueError: your out spatial shape [0, 31, 27] reach zero!!! input shape: [1, 62, 54]
```
For **pointrcnn** it is another error:
`  File "/home/ganoufa/workSpace/deep_learning/lidar_detection/OpenPCDet/tools/../pcdet/models/backbones_3d/pointnet2_backbone.py", line 76, in forward
    assert xyz_batch_cnt.min() == xyz_batch_cnt.max()
AssertionError
`

closed 2022-06-11 by github-actions[bot]

GBagley, 2022-08-29:
Anyone figure this out? I'm having the same issue

pk1996, 2022-09-05:
Try playing with the voxel size. It dictates the gird size and it is not big enough for the different downsamples happening in different layers!

meanmee, 2023-06-14:
any progress on this issue?

meanmee, 2023-06-14:
I just found that if the .bin file had no data, then the error might occur