open-mmlab/OpenPCDet #954: Explanation points_in_boxes_kernel Function

antoniosilva9116, 2022-05-03:
I have a question regarding cuda function in file roiaware_pool3d_kernel.cu. In this function, we have the following code:
```
__global__ void points_in_boxes_kernel(int batch_size, int boxes_num, int pts_num, const float *boxes,
    const float *pts, int *box_idx_of_points){

      int bs_idx = blockIdx.y;
      (...)
      
      boxes += bs_idx * boxes_num * 7;
      (...)
}
```
So, my question is about how can we make an operation (summing) with an array (boxes: (B, N, 7) [x, y, z, dx, dy, dz, heading]) and a float value?

sshaoshuai, 2022-05-03:
Here the boxes is a pointer (point to some memory address) and the "+=" operation aims to shift the pointer to some new position for this sample (bs_idx).

antoniosilva9116, 2022-05-03:
How could I translate this shift into python? do you have any idea?

closed 2022-06-17 by github-actions[bot]