open-mmlab/OpenPCDet #960: Optimizer neglection when using nn.Parameter and 'adam_one_cycle'

xiaoxin83121, 2022-05-08:
Hi, thanks to your great work!
I have noticed when building optimizer in 'adam_one_cycle' manner, only parameters in the leaf module are added to the optimizer parameter groups. I have tried a `nn.Parameter()`  in a non-leaf module, but it is not optimized since the building implementation in [line 26~32](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/train_utils/optimization/__init__.py)

I have solve it by adding a wrap module. Is there any fix plan about that or better solution?

closed 2022-06-22 by github-actions[bot]

huixiancheng, 2023-07-17:
Same problem. Would you like to share your resolution. Thanks a lot.

xiaoxin83121, 2023-08-10:
> Same problem. Would you like to share your resolution. Thanks a lot.

hi, just wrap `nn.Parameters` with a Subclass of `nn.Module`, and return it in `forward`