open-mmlab/OpenPCDet #962: There was an error in infering the model  on jetson NX2

gzchenjiajun, 2022-05-09:
2022-05-09 20:49:07,460   INFO  ==> Loading parameters from checkpoint ./checkpoint_epoch_2.pth to GPU
2022-05-09 20:49:07,671   INFO  ==> Checkpoint trained from version: pcdet+0.5.2+c533261
2022-05-09 20:49:07,796   INFO  Not updated weight backbone_3d.conv_input.0.weight: torch.Size([3, 3, 3, 4, 16])
2022-05-09 20:49:07,796   INFO  Not updated weight backbone_3d.conv1.0.0.weight: torch.Size([3, 3, 3, 16, 16])
2022-05-09 20:49:07,796   INFO  Not updated weight backbone_3d.conv2.0.0.weight: torch.Size([3, 3, 3, 16, 32])
2022-05-09 20:49:07,797   INFO  Not updated weight backbone_3d.conv2.1.0.weight: torch.Size([3, 3, 3, 32, 32])
2022-05-09 20:49:07,797   INFO  Not updated weight backbone_3d.conv2.2.0.weight: torch.Size([3, 3, 3, 32, 32])
2022-05-09 20:49:07,797   INFO  Not updated weight backbone_3d.conv3.0.0.weight: torch.Size([3, 3, 3, 32, 64])
2022-05-09 20:49:07,797   INFO  Not updated weight backbone_3d.conv3.1.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-09 20:49:07,797   INFO  Not updated weight backbone_3d.conv3.2.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-09 20:49:07,798   INFO  Not updated weight backbone_3d.conv4.0.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-09 20:49:07,798   INFO  Not updated weight backbone_3d.conv4.1.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-09 20:49:07,798   INFO  Not updated weight backbone_3d.conv4.2.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-09 20:49:07,798   INFO  Not updated weight backbone_3d.conv_out.0.weight: torch.Size([3, 1, 1, 64, 128])
2022-05-09 20:49:07,798   INFO  ==> Done (loaded 151/163)

The self-trained model reasoned well on x86 device, but failed on Jetson NX2.
The not updated weight backbone...

how to solve?

jetson :torch 1.6
x86 :torch 1.4

Does it have anything to do with the Torch version?

gzchenjiajun, 2022-05-09:
Bonus: The official pre-training model is normal on the Jetson NX2

doubisu, 2022-06-11:
I had got same error. For me, the cause was difference of spconv version. When I use same spconv (and cumm) version at learning machine and jetson, the error  disappeared.

closed 2022-07-25 by github-actions[bot]