open-mmlab/OpenPCDet #967: Problems with trained model with waymo dataset

udnqv, 2022-05-11:
Hi,
I trained a pv_rcnn++ model with the waymo dataset. Sadly when I try to use demo.py with the trained model I get following error message:
[image]
There is a possibilty that I overlook something very easy, because I'm very new to this. Also I didn't quite understand what in the Getting Started was mean by "transform your point cloud data" I did define point[:, 3]= 0 but that's it. Is there something else I have to do? I didn't change anything else in the whole repo also. If anyone has an idea or faced a similar issue and knows a solution, please share below. 
Thanks in advance.

aldipiroli, 2022-05-23:
The issue might be that the number of features that your input point cloud has is different from the one that the DataProcessor modules expects. 
For the waymo dataset, the expected features are defined in waymo_dataset.yaml
```
POINT_FEATURE_ENCODING:
  {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ["x", "y", "z", "intensity", "elongation"],
    src_feature_list: ["x", "y", "z", "intensity", "elongation"],
  }
```
so in total you should have 5 features.
You can try and give as input points something like this: `points = torch.rand((1000, 5))` to see if that is really the problem.

closed 2022-07-08 by github-actions[bot]

jinhuilin97, 2022-09-29:
I got the same problem .
and Whether I can test on other datasets?
how

jinhuilin97, 2022-09-30:
[image]
got it ，elongation -> padding zero can solve it 
but I don't know how to generate elongation