open-mmlab/OpenPCDet #970: How to add score to demo output

Leozyc-waseda, 2022-05-17:
https://github.com/open-mmlab/OpenPCDet/blob/846cf3ed21c274f95f041e7e21fd039ca5c99a1f/tools/visual_utils/open3d_vis_utils.py#L113-L115

I found that there is no add 3d label function in open3d.visualization.Visualizer, I am curious how to got score output.

shaoxh, 2022-05-19:
This function is no longer avaliable in open3d. You may refer to codes below to make 3d label.
That's a lot to change.
For details go to [open3d](https://github.com/isl-org/Open3D/blob/master/examples/python/visualization/text3d.py)
```
def low_level():
    app = gui.Application.instance
    app.initialize()

    points = make_point_cloud(100, (0, 0, 0), 1.0)

    w = app.create_window("Open3D - 3D Text", 1024, 768)
    widget3d = gui.SceneWidget()
    widget3d.scene = rendering.Open3DScene(w.renderer)
    mat = rendering.MaterialRecord()
    mat.shader = "defaultUnlit"
    mat.point_size = 5 * w.scaling
    widget3d.scene.add_geometry("Points", points, mat)
    for idx in range(0, len(points.points)):
        l = widget3d.add_3d_label(points.points[idx], "{}".format(idx))
        l.color = gui.Color(points.colors[idx][0], points.colors[idx][1],
                            points.colors[idx][2])
        l.scale = np.random.uniform(0.5, 3.0)
    bbox = widget3d.scene.bounding_box
    widget3d.setup_camera(60.0, bbox, bbox.get_center())
    w.add_child(widget3d)

    app.run()
```

Leozyc-waseda, 2022-05-20:
or change 
`open3d.visualization.Visualizer` to `open3d.visualization.O3DVisualizer` to continue use add 3d label function.
Still a lot to change.
and also OpenGL >=4.. 
I don't know yet how to upgrade my OpenGl :(

jihanyang, 2022-05-24:
I comment this code since this function is not supported by open3d 0.9.0, which means that not all environments using the openpcdet can use this function. You can use mayavi for visualization with text in existing openpcdet released code.

closed 2022-05-24 by Leozyc-waseda

Petros626, 2025-06-04:
> O3DVisualizer

can you provide a running example with adapted `open3d_vis_utils.py` for it? I struggle with the new visualizer

Petros626, 2025-09-13:
I finally used a function copied from `Open3D` issues:

[image]