open-mmlab/OpenPCDet #973: create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset11.yaml

hanjiangrui, 2022-05-18:
when i creat kitti data info, code run last date 
        with futures.ThreadPoolExecutor(num_workers) as executor:
            infos = executor.map(process_single_scene, sample_id_list)
        return list(infos)
Error:
Traceback (most recent call last):
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 586, in result_iterator
    yield fs.pop().result()
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 425, in result
    return self.__get_result()
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/data2/project/DL_Demo/OpenPCDet/pcdet/datasets/kitti/kitti_dataset.py", line 183, in process_single_scene
    annotations['bbox'] = np.concatenate([obj.box2d.reshape(1, 4) for obj in obj_list], axis=0)
  File "<__array_function__ internals>", line 6, in concatenate
ValueError: need at least one array to concatenate

closed 2022-07-02 by github-actions[bot]

ye-sama, 2022-09-01:
老哥搞好了吗

xpyqiubai, 2022-09-08:
I have same question about this.