open-mmlab/OpenPCDet #978: Can't see installed pcdet pip package

luigifaticoso, 2022-05-23:
Hello everyone, 

After the installation of the pcdet pip package I can't find it. When I type `pip list` i can't see `pcdet`. I have followed the installations steps locally and even using the docker image but don't seem to be working for me. As soon as i run the demo it says `ModuleNotFoundError: No module named 'pcdet'` 

How can i solve this issue? 

This is the output of my `python setup.py develop` that seems to end correctly: 

```
/root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp: In function ‘int ball_query_wrapper_fast(int, int, int, float, int, at::Tensor, at::Tensor, at::Tensor)’:
/root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:18:16: warning: ‘at::DeprecatedTypeProperties& at::Tensor::type() const’ is deprecated: Tensor.type() is deprecated. Instead use Tensor.options(), which in many cases (e.g. in a constructor) is a drop-in replacement. If you were using data from type(), that is now available from Tensor itself, so instead of tensor.type().scalar_type(), use tensor.scalar_type() instead and instead of tensor.type().backend() use tensor.device(). [-Wdeprecated-declarations]
    if (!x.type().is_cuda()) { \
                ^
/root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:29:24: note: in expansion of macro ‘CHECK_CUDA’
 #define CHECK_INPUT(x) CHECK_CUDA(x);CHECK_CONTIGUOUS(x)
                        ^~~~~~~~~~
/root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:34:5: note: in expansion of macro ‘CHECK_INPUT’
     CHECK_INPUT(new_xyz_tensor);
     ^
In file included from /root/.local/lib/python3.7/site-packages/torch/include/ATen/Tensor.h:3:0,
                 from /root/.local/lib/python3.7/site-packages/torch/include/ATen/Context.h:4,
                 from /root/.local/lib/python3.7/site-packages/torch/include/ATen/ATen.h:5,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from /root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:8:
/root/.local/lib/python3.7/site-packages/torch/include/ATen/core/TensorBody.h:268:30: note: declared here
   DeprecatedTypeProperties & type() const {
                              ^~~~
/root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:18:16: warning: ‘at::DeprecatedTypeProperties& at::Tensor::type() const’ is deprecated: Tensor.type() is deprecated. Instead use Tensor.options(), which in many cases (e.g. in a constructor) is a drop-in replacement. If you were using data from type(), that is now available from Tensor itself, so instead of tensor.type().scalar_type(), use tensor.scalar_type() instead and instead of tensor.type().backend() use tensor.device(). [-Wdeprecated-declarations]
    if (!x.type().is_cuda()) { \
                ^
/root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:29:24: note: in expansion of macro ‘CHECK_CUDA’
 #define CHECK_INPUT(x) CHECK_CUDA(x);CHECK_CONTIGUOUS(x)
                        ^~~~~~~~~~
/root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:35:5: note: in expansion of macro ‘CHECK_INPUT’
     CHECK_INPUT(xyz_tensor);
     ^
In file included from /root/.local/lib/python3.7/site-packages/torch/include/ATen/Tensor.h:3:0,
                 from /root/.local/lib/python3.7/site-packages/torch/include/ATen/Context.h:4,
                 from /root/.local/lib/python3.7/site-packages/torch/include/ATen/ATen.h:5,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from /root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:8:
/root/.local/lib/python3.7/site-packages/torch/include/ATen/core/TensorBody.h:268:30: note: declared here
   DeprecatedTypeProperties & type() const {
                              ^~~~
/root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:36:55: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
     const float *new_xyz = new_xyz_tensor.data<float>();
                                                       ^
In file included from /root/.local/lib/python3.7/site-packages/torch/include/ATen/Tensor.h:3:0,
                 from /root/.local/lib/python3.7/site-packages/torch/include/ATen/Context.h:4,
                 from /root/.local/lib/python3.7/site-packages/torch/include/ATen/ATen.h:5,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /root/.local/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from /root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:8:
/root/.local/lib/python3.7/site-packages/torch/include/ATen/core/TensorBody.h:354:7: note: declared here
   T * data() const {
       ^~~~
/root/OpenPCDet/pcdet/ops/pointnet2/pointnet2_batch/src/ball_query.cpp:37:47: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
     const float *xyz = xyz_tensor.data<float>();
                                               ^
In file included from /root/.local/lib/python3.7/site-packages/torch/include/ATen/Tensor.h:3:0,
                 from /root/.local/lib/python3.7/site-packages/torch/include/ATen/Context.h:4,
                 from /root/.local/lib/python3.7/site-packages/torch/include/ATen/ATen.h:5,
[... 344 lines cut ...]
portalocker        2.4.0
protobuf           3.20.1
pybind11           2.9.2
pyparsing          3.0.9
PyWavelets         1.3.0
PyYAML             6.0
requests           2.27.1
scikit-image       0.19.2
scipy              1.7.3
setuptools         62.3.2
SharedArray        3.2.1
six                1.16.0
spconv-cu102       2.1.21
tensorboardX       2.5
termcolor          1.1.0
tifffile           2021.11.2
torch              1.6.0
torchvision        0.7.0
tqdm               4.64.0
typing_extensions  4.2.0
urllib3            1.26.9
wheel              0.37.1
zipp               3.8.0
```
Thank you

K3vwho, 2022-05-27:
Add 'import _init_path' before the imports of pcdet in demo.py worked for me.

qiliang72, 2022-06-01:
> Add 'import _init_path' before the imports of pcdet in demo.py worked for me.

Thanks so much! This also worked for me.

closed 2022-07-17 by github-actions[bot]

shadismz, 2022-09-03:
> Hello everyone,
>
> After the installation of the pcdet pip package I can't find it. When I type `pip list` i can't see `pcdet`. I have followed the installations steps locally and even using the [...]

@lindahua Hello,
Did you solve this problem?
I have the same problem. and I tried adding `import _init_path` befor pcdet errors in demp.py, but still got the same error.
would you please say how did you solve it?

IIashaash, 2024-04-18:
same problem didn't solve still gave same error