open-mmlab/OpenPCDet #979: Infer model was not loaded normally on the Jetson TX2 development board

gzchenjiajun, 2022-05-24:
2022-05-24 10:48:17,167   INFO  ==> Loading parameters from checkpoint ./checkpoint_epoch_1.pth to GPU
2022-05-24 10:48:21,515   INFO  ==> Checkpoint trained from version: pcdet+0.5.2+c533261+py48dbe0c
2022-05-24 10:48:21,631   INFO  Not updated weight backbone_3d.conv_input.0.weight: torch.Size([3, 3, 3, 4, 16])
2022-05-24 10:48:21,631   INFO  Not updated weight backbone_3d.conv1.0.0.weight: torch.Size([3, 3, 3, 16, 16])
2022-05-24 10:48:21,632   INFO  Not updated weight backbone_3d.conv2.0.0.weight: torch.Size([3, 3, 3, 16, 32])
2022-05-24 10:48:21,632   INFO  Not updated weight backbone_3d.conv2.1.0.weight: torch.Size([3, 3, 3, 32, 32])
2022-05-24 10:48:21,632   INFO  Not updated weight backbone_3d.conv2.2.0.weight: torch.Size([3, 3, 3, 32, 32])
2022-05-24 10:48:21,632   INFO  Not updated weight backbone_3d.conv3.0.0.weight: torch.Size([3, 3, 3, 32, 64])
2022-05-24 10:48:21,632   INFO  Not updated weight backbone_3d.conv3.1.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-24 10:48:21,633   INFO  Not updated weight backbone_3d.conv3.2.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-24 10:48:21,633   INFO  Not updated weight backbone_3d.conv4.0.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-24 10:48:21,633   INFO  Not updated weight backbone_3d.conv4.1.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-24 10:48:21,633   INFO  Not updated weight backbone_3d.conv4.2.0.weight: torch.Size([3, 3, 3, 64, 64])
2022-05-24 10:48:21,633   INFO  Not updated weight backbone_3d.conv_out.0.weight: torch.Size([3, 1, 1, 64, 128])
2022-05-24 10:48:21,634   INFO  ==> Done (loaded 245/257)
test.pcd

Infer model was not loaded normally on the Jetson TX2 development board
how to solve?

gzchenjiajun, 2022-05-24:
his happens to all of second,pointpillar,voxel_rcnn_car...

jihanyang, 2022-05-25:
Could you make sure that this also happen in pointpillar?

closed 2022-07-10 by github-actions[bot]