open-mmlab/OpenPCDet #982: PVRCNN tested on my dataset, got a lot of false postive

Leozyc-waseda, 2022-05-26:
How to fix this problem?

I am using PVRCNN to test on my 32 beam dataset, pretrained by Pandaset 64 beam.

[image]

udacityyy, 2022-05-30:
Hello, sorry to ask this, but could you tell me which tool did you use to visualize the results?

luminous33, 2022-05-31:
> How to fix this problem?
> 
> I am using PVRCNN to test on my 32 beam dataset, pretrained by Pandaset 64 beam.
> 
> [image]

Hello if you are ok, Could i get pretrained by Pandaset 64 beam?
zmfkvlzk@gmail.com -> this is my email address.

I tried train own my dataset. But it keeps failing.
If you can help me, it will be a great help to me :)

Thk

OrangeSodahub, 2022-07-18:
Not so sure about what problem you may have. But can refer to my successful achievement on training my own dataset and inference correctly https://github.com/OrangeSodahub/CRLFnet

closed 2022-09-02 by github-actions[bot]