open-mmlab/OpenPCDet #987: I need my custom_data training without 2d infos, also i dont want downgrade

luminous33, 2022-05-30:
<details><summary>pip env</summary>
<p>
absl-py                     0.15.0
addict                      2.4.0
anyio                       3.5.0
argon2-cffi                 21.3.0
argon2-cffi-bindings        21.2.0
asttokens                   2.0.5
astunparse                  1.6.3
attrs                       21.4.0
Babel                       2.10.1
backcall                    0.2.0
beautifulsoup4              4.11.1
black                       22.3.0
bleach                      5.0.0
cachetools                  5.0.0
ccimport                    0.3.7
certifi                     2021.10.8
cffi                        1.15.0
charset-normalizer          2.0.12
click                       8.1.3
colorama                    0.4.4
cumm                        0.2.8
cumm-cu113                  0.2.8
cycler                      0.11.0
Cython                      0.29.28
debugpy                     1.6.0
decorator                   5.1.1
defusedxml                  0.7.1
deprecation                 2.1.0
descartes                   1.1.0
dgl                         0.6.1
easydict                    1.9
entrypoints                 0.4
executing                   0.8.3
fastjsonschema              2.15.3
filelock                    3.6.0
fire                        0.4.0
flake8                      4.0.1
flatbuffers                 1.12
fonttools                   4.33.3
gast                        0.3.3
google-auth                 2.6.6
google-auth-oauthlib        0.4.6
google-pasta                0.2.0
grpcio                      1.32.0
h5py                        2.10.0
idna                        3.3
imageio                     2.19.0
importlib-metadata          4.11.3
importlib-resources         5.7.1
iniconfig                   1.1.1
ipykernel                   6.13.0
ipython                     8.3.0
ipython-genutils            0.2.0
ipywidgets                  7.7.0
jedi                        0.18.1
Jinja2                      3.1.2
joblib                      1.1.0
json5                       0.9.6
jsonschema                  4.4.0
jupyter                     1.0.0
jupyter-client              7.3.0
jupyter-console             6.4.3
jupyter-core                4.10.0
jupyter-packaging           0.12.0
jupyter-server              1.17.0
jupyterlab                  3.3.4
jupyterlab-pygments         0.2.2
jupyterlab-server           2.13.0
jupyterlab-widgets          1.1.0
keras-nightly               2.5.0.dev2021032900
Keras-Preprocessing         1.1.2
kiwisolver                  1.4.2
lark                        1.1.2
llvmlite                    0.36.0
lyft-dataset-sdk            0.0.8
Markdown                    3.3.6
MarkupSafe                  2.1.1
matplotlib                  3.5.1
matplotlib-inline           0.1.3
mccabe                      0.6.1
mistune                     0.8.4
mmcls                       0.23.0
mmcv-full                   1.5.0
mmdet                       2.24.1
mmdet3d                     1.0.0rc2            /data/test_model/mm/OpenPCDet/mmdetection3d
mmsegmentation              0.24.1
model-index                 0.1.11
mypy-extensions             0.4.3
nbclassic                   0.3.7
nbclient                    0.6.0
nbconvert                   6.5.0
nbformat                    5.3.0
nest-asyncio                1.5.5
networkx                    2.2
ninja                       1.10.2.3
nose                        1.3.7
notebook                    6.4.11
notebook-shim               0.1.0
numba                       0.53.0
numpy                       1.21.1
nuscenes-devkit             1.1.9
oauthlib                    3.2.0
olefile                     0.46
open3d                      0.15.2
opencv-python               4.5.5.64
opt-einsum                  3.3.0
ordered-set                 4.1.0
packaging                   21.3
pandas                      1.4.2
pandocfilters               1.5.0
parso                       0.8.3
pathspec                    0.9.0
pccm                        0.3.4
pcdet                       0.5.2+846cf3e       /data/test_model/mm/OpenPCDet
pexpect                     4.8.0
pickleshare                 0.7.5
Pillow                      9.1.0
pip                         21.2.4
platformdirs                2.5.2
plotly                      5.7.0
pluggy                      1.0.0
plyfile                     0.7.4
portalocker                 2.4.0
prettytable                 3.3.0
prometheus-client           0.14.1
prompt-toolkit              3.0.29
protobuf                    3.9.2
psutil                      5.9.0
ptyprocess                  0.7.0
pure-eval                   0.2.2
py                          1.11.0
pyasn1                      0.4.8
pyasn1-modules              0.2.8
pybind11                    2.9.2
pycocotools                 2.0.4
pycodestyle                 2.8.0
pycparser                   2.21
pyflakes                    2.4.0
Pygments                    2.12.0
pyparsing                   3.0.8
pypcd                       0.1.1
pyquaternion                0.9.9
pyrsistent                  0.18.1
pytest                      7.1.2
python-dateutil             2.8.2
python-lzf                  0.2.4
pytz                        2022.1
PyWavelets                  1.3.0
PyYAML                      6.0
pyzmq                       22.3.0
qtconsole                   5.3.0
QtPy                        2.1.0
requests                    2.27.1
requests-oauthlib           1.3.1
rsa                         4.8
scikit-image                0.19.2
scikit-learn                1.0.2
scipy                       1.8.0
Send2Trash                  1.8.0
setuptools                  61.2.0
Shapely                     1.8.2
SharedArray                 3.1.0
six                         1.15.0
sniffio                     1.2.0
soupsieve                   2.3.2.post1
spconv                      2.1.21
spconv-cu113                2.1.21
stack-data                  0.2.0
tabulate                    0.8.9
tenacity                    8.0.1
tensorboard-data-server     0.6.1
tensorboard-plugin-wit      1.8.1
tensorboardX                2.5
tensorflow                  2.4.0
tensorflow-estimator        2.4.0
termcolor                   1.1.0
terminado                   0.13.3
terminaltables              3.1.10
threadpoolctl               3.1.0
tifffile                    2022.4.28
tinycss2                    1.1.1
tomli                       2.0.1
tomlkit                     0.10.2
torch                       1.9.1
torchaudio                  0.9.0
torchvision                 0.10.1
tornado                     6.1
tqdm                        4.64.0
traitlets                   5.1.1
trimesh                     2.35.39
typing_extensions           4.2.0
urllib3                     1.26.9
waymo-open-dataset-tf-2-4-0 1.4.1
wcwidth                     0.2.5
webencodings                0.5.1
websocket-client            1.3.2
Werkzeug                    2.1.2
wheel                       0.37.1
widgetsnbextension          3.6.0
wrapt                       1.12.1
yapf                        0.32.0
zipp                        3.8.0
</p>
</details>

I already installed every envs.

when i tried demo.py,
`Traceback (most recent call last):
  File "demo.py", line 112, in <module>
    main()
  File "demo.py", line 98, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/data/test_model/mm/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/usr/local/anaconda3/envs/mmdet3d/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/test_model/mm/OpenPCDet/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 352, in forward
    keypoints = self.get_sampled_points(batch_dict)
  File "/data/test_model/mm/OpenPCDet/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 254, in get_sampled_points
    cur_pt_idxs = pointnet2_stack_utils.farthest_point_sample(
  File "/data/test_model/mm/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/pointnet2_utils.py", line 180, in forward
    pointnet2.farthest_point_sampling_wrapper(B, N, npoint, xyz, temp, output)
AttributeError: module 'pcdet.ops.pointnet2.pointnet2_stack.pointnet2_stack_cuda' has no attribute 'farthest_point_sampling_wrapper'
`
IDK :(

Please you can help me.

And one more question
I want to train without 2d images, calibration, 2d label(2d infos)
i tried make own_datsets.py
but it doesnt work.

Any comments you can give me please.

Thk :)

sshaoshuai, 2022-06-03:
Please try to recompile the repo with `python setup.py clean` and `python setup.py develop` under the directory of OpenPCDet.

luminous33, 2022-06-07:
> Please try to recompile the repo with `python setup.py clean` and `python setup.py develop` under the directory of OpenPCDet.

same issue.
it only happens when i do only pointrcnn. When I run other models likes pvrcnn, second etc. it works fine.

Additionally Could you tell me how to train without 2d information(images, calibration, 2d coordinate etc)

Thank you for your attention to this matter :)

jihanyang, 2022-06-07:
There are already massive discussion about custom dataset without 2d information. Besides, the implemented waymo dataset and nuscenes dataset also can work without images. Maybe those discussion and code could help you.

closed 2022-07-23 by github-actions[bot]