open-mmlab/OpenPCDet #995: Where the code to transform the original nuscene lidar point clouds to unified normative coordinate?

ideasplus, 2022-06-07:
Hi

The coordinate of original nuscene lidar point clouds is **x pointing right, y pointing forward**. However, I can't find the code to explicitly transform the nuscene lidar coordinate to the unified normative coordinate as the documentation states. 

The __getitem__ method of nuscene

https://github.com/open-mmlab/OpenPCDet/blob/dadda9ed5c0f190d06856c6ec15d552116203947/pcdet/datasets/nuscenes/nuscenes_dataset.py#L117-L151

The get_sweep method of nuscene
https://github.com/open-mmlab/OpenPCDet/blob/dadda9ed5c0f190d06856c6ec15d552116203947/pcdet/datasets/nuscenes/nuscenes_dataset.py#L76-L109

lihao0374, 2022-07-05:
same question...

closed 2022-08-19 by github-actions[bot]

OrangeSodahub, 2023-02-05:
@ideasplus @lihao0374 Have you solved?

Petros626, 2026-02-12:
@OrangeSodahub it seems not, so can you clarify?