open-mmlab/OpenPCDet #997: How to obtain inference speed？

VsionQing, 2022-06-12:
How to calculate the inference speed FPs in this paper?
PV-RCNN and so on

Jane-QinJ, 2022-06-24:
Have the same question

Jane-QinJ, 2022-06-30:
> We use `time.time()` with `os.environ['CUDA_LAUNCH_BLOCKING'] = '1'`

Thank u for your gentle reply! But I'm not familiar enough to your great work. So could u gentlely tell me in detail how to write this code and use it in your OpenPCDet framework? I'll be very appreciated to your instruction.

strugglingguoguo, 2022-07-22:
I Have the same question.I don't know where to put the time module.

jihanyang, 2022-07-22:
Put `time.time()` here to calculate the forward time: 
https://github.com/open-mmlab/OpenPCDet/blob/c233477a4a4bf2861a978794d6b9c047f64641f7/tools/eval_utils/eval_utils.py#L56-L57

jihanyang, 2022-08-21:
Supported in the latest verison with `--infer_time` keyword

closed 2022-08-21 by jihanyang

ammaryasirnaich, 2023-07-27:
Is there a difference between --infer_time and FPS ?