open-mmlab/OpenPCDet #1002: Installation. Ninja Error

Tottowich, 2022-06-14:
I'm trying to set up the but keep getting this extensive error message which i can't seem to resolve.
I have CUDA 11.4 installed on my system and the following venv:
(venv) C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet>pip list 
Package            Version
------------------ ------------
ccimport           0.3.7
certifi            2022.5.18.1
charset-normalizer 2.0.12
cumm-cu114         0.2.8
easydict           1.9
fire               0.4.0
idna               3.3
lark               1.1.2
ninja              1.10.2.3
numpy              1.22.4
pccm               0.3.4
Pillow             9.1.1
pip                22.1.2
portalocker        2.4.0
pybind11           2.9.2
pywin32            304
requests           2.28.0
setuptools         62.2.0
six                1.16.0
spconv-cu114       2.1.22
termcolor          1.1.0
tifffile           2022.5.4
torch              1.10.1+cu113
torchvision        0.11.2+cu113
typing_extensions  4.2.0
urllib3            1.26.9
wheel              0.37.1
I'm on windows with a 3060Ti.
The error message occurs when running
       python setup.py develop
python == 3.9.13
C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\include\c10/util/Optional.h(383): note: see reference to class template instantiation 'c10::constexpr_storage_t<T>' being compiled
        with
        [
            T=std::unordered_map<std::string,c10::IValue,std::hash<std::string>,std::equal_to<std::string>,std::allocator<std::pair<const std::string,c10::IValue>>>
        ]
C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\include\c10/util/Optional.h(526): note: see reference to class template instantiation 'c10::trivially_copyable_optimization_optional_base<T>' being compiled
        with
        [
            T=std::unordered_map<std::string,c10::IValue,std::hash<std::string>,std::equal_to<std::string>,std::allocator<std::pair<const std::string,c10::IValue>>>
        ]
C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\include\c10/util/Optional.h(526): note: see reference to alias template instantiation 'c10::OptionalBase<T>' 
being compiled
        with
        [
            T=std::unordered_map<std::string,c10::IValue,std::hash<std::string>,std::equal_to<std::string>,std::allocator<std::pair<const std::string,c10::IValue>>>
        ]
C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\include\torch/csrc/autograd/profiler_kineto.h(48): note: see reference to class template instantiation 'c10::optional<std::unordered_map<std::string,c10::IValue,std::hash<std::string>,std::equal_to<std::string>,std::allocator<std::pair<const std::string,c10::IValue>>>>' being compiled
C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\include\c10/util/Optional.h(418): warning C4624: 'c10::trivially_copyable_optimization_optional_base<T>': destructor was implicitly defined as deleted
        with
        [
            T=std::unordered_map<std::string,c10::IValue,std::hash<std::string>,std::equal_to<std::string>,std::allocator<std::pair<const std::string,c10::IValue>>>
        ]
C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\include\c10/util/Optional.h(184): warning C4624: 'c10::constexpr_storage_t<T>': destructor was implicitly defined as deleted
        with
        [
            T=std::vector<at::Tensor,std::allocator<at::Tensor>>
        ]
C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\include\c10/util/Optional.h(383): note: see reference to class template instantiation 'c10::constexpr_storage_t<T>' being compiled
        with
        [
            T=std::vector<at::Tensor,std::allocator<at::Tensor>>
        ]
C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\include\c10/util/Optional.h(526): note: see reference to class template instantiation 'c10::trivially_copyable_optimization_optional_base<T>' being compiled
        with
[... 77 lines cut ...]
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\setuptools\dist.py", line 1224, in run_command
    super().run_command(command)
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\setuptools\_distutils\dist.py", line 986, in run_command
    cmd_obj.run()
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\setuptools\command\build_ext.py", line 79, in run
    _build_ext.run(self)
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 339, in run
    self.build_extensions()
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\utils\cpp_extension.py", line 735, in build_extensions
    build_ext.build_extensions(self)
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\setuptools\command\build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\utils\cpp_extension.py", line 708, in win_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\utils\cpp_extension.py", line 1399, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\torch\utils\cpp_extension.py", line 1733, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension

Tottowich, 2022-06-14:
Noticed that SharedArray was not installed. But when calling:
pip install SharedArray ->
(env) C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet>pip install SharedArray
Collecting SharedArray
  Using cached SharedArray-3.2.1.tar.gz (22 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy in c:\users\theodor jonsson\visual_studio_code\machine_learning\openpcdet\env\lib\site-packages (from SharedArray) (1.22.4)
Building wheels for collected packages: SharedArray
  Building wheel for SharedArray (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
      running bdist_wheel
      running build
      running build_ext
      building 'SharedArray' extension
      creating build
      creating build\temp.win-amd64-cpython-39
      creating build\temp.win-amd64-cpython-39\Release
      creating build\temp.win-amd64-cpython-39\Release\src
      "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD "-IC:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\numpy\core\include" "-IC:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\include" "-IC:\Program Files\Python39\include" "-IC:\Program Files\Python39\Include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\cppwinrt" /Tc.\src\map_owner.c /Fobuild\temp.win-amd64-cpython-39\Release\.\src\map_owner.obj
      map_owner.c
      .\src\map_owner.c(19): fatal error C1083: Cannot open include file: 'sys/mman.h': No such file or directory
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.32.31326\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for SharedArray
  Running setup.py clean for SharedArray
Failed to build SharedArray
Installing collected packages: SharedArray
  Running setup.py install for SharedArray ... error
  error: subprocess-exited-with-error

  × Running setup.py install for SharedArray did not run successfully.
  │ exit code: 1
  ╰─> [14 lines of output]
      running install
      C:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_ext
      building 'SharedArray' extension
      creating build
      creating build\temp.win-amd64-cpython-39
      creating build\temp.win-amd64-cpython-39\Release
      creating build\temp.win-amd64-cpython-39\Release\src
      "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD "-IC:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\lib\site-packages\numpy\core\include" "-IC:\Users\Theodor Jonsson\Visual_studio_code\Machine_learning\OpenPCDet\env\include" "-IC:\Program Files\Python39\include" "-IC:\Program Files\Python39\Include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\cppwinrt" /Tc.\src\map_owner.c /Fobuild\temp.win-amd64-cpython-39\Release\.\src\map_owner.obj
      map_owner.c
      .\src\map_owner.c(19): fatal error C1083: Cannot open include file: 'sys/mman.h': No such file or directory
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.32.31326\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> SharedArray

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

siddas27, 2022-07-04:
I am facing the same issue. Did you find any solution?

ZecCheng, 2022-07-09:
open-mmlab/OpenPCDet$ python setup.py develop
running develop
/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/setuptools/command/easy_install.py:147: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  EasyInstallDeprecationWarning,
/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/utils/cpp_extension.py:788: UserWarning: The detected CUDA version (11.5) has a minor version mismatch with the version that was used to compile PyTorch (11.3). Most likely this shouldn't be a problem.
  warnings.warn(CUDA_MISMATCH_WARN.format(cuda_str_version, torch.version.cuda))
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
Emitting ninja build file /opt/sdatmp/lcx/anaconda/envs/open-mmlab/OpenPCDet/build/temp.linux-x86_64-3.7/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/1] /usr/local/cuda-11.5/bin/nvcc  -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/include -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/include/TH -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda-11.5/include -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/include/python3.7m -c -c /opt/sdatmp/lcx/anaconda/envs/open-mmlab/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu -o /opt/sdatmp/lcx/anaconda/envs/open-mmlab/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
FAILED: /opt/sdatmp/lcx/anaconda/envs/open-mmlab/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o
/usr/local/cuda-11.5/bin/nvcc  -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/include -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/include/TH -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda-11.5/include -I/opt/sdatmp/lcx/anaconda/envs/open-mmlab/include/python3.7m -c -c /opt/sdatmp/lcx/anaconda/envs/open-mmlab/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu -o /opt/sdatmp/lcx/anaconda/envs/open-mmlab/OpenPCDet/build/temp.linux-x86_64-3.7/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
/usr/include/stdio.h(183): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(195): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(217): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(254): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(279): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(288): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(297): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(303): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(309): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(824): error: attribute "__malloc__" does not take arguments

/usr/include/stdlib.h(565): error: attribute "__malloc__" does not take arguments

/usr/include/stdlib.h(569): error: attribute "__malloc__" does not take arguments

/usr/include/stdlib.h(797): error: attribute "__malloc__" does not take arguments

13 errors detected in the compilation of "/opt/sdatmp/lcx/anaconda/envs/open-mmlab/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu".
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
[... cut ...]
  File "/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1424, in _write_ninja_file_and_compile_objects
    error_prefix='Error compiling objects for extension')
  File "/opt/sdatmp/lcx/anaconda/envs/open-mmlab/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1756, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension

I am facing the same issue. Did you find any solution?

leducnguyen21, 2022-07-09:
@ZecCheng I also got the same problem. What CUDA and Pytorch are you using, I think the issue can come from the compatible of Pytorch and CUDA are not matched

Tottowich, 2022-07-12:
Sorry for the late reply but I could not seem to find any solutions. I dual booted my PC and ran via Linux. You could try looking at this issue https://github.com/open-mmlab/OpenPCDet/issues/681, #681. It gives some suggestions on how one might change the cpp files in order to run it on windows. The main issue is that SpConv does not run on Win but if you use Pointpillars it doesn't use that library so you could also try that.

closed 2022-08-26 by github-actions[bot]